bambulab/BambuStudio #4864: Format error (overlapping of output fields) in the display (window) of the estimated printing time for several printing plates.

Ringo-092024, 2024-09-21:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Windows 10 Pro / Version 22H

### Additional system information

Intel(R) Core(TM) i7-4820K CPU @ 3.70GHz   3.70 GHz 
64 GB RAM
512 GB SSD (C:)
NVIDIA GeForce GTX 780
Monitor 27" (2560x1440)

### Printer

Bambu Lab X1C

### How to reproduce

1. Open File "V-Spooler+1.0+-+Bambu.3mf" , source Maker World (https://makerworld.com/de/models/98832#profileId-105442)
2. then Preview 
3. Slice all printing plates
4. display all printing times all printing plate as overview of all printing plates in separate small window

[image] 

### Actual results

Everything below is without errors.

### Expected results

Reformatting the output window.

[image]

### Project file & Debug log uploads

V-Spooler+1.0+-+Bambu.3mf (40MB)
(Error-Message: File size too big: 25 MB are allowed, 37 MB were attempted to upload.V-Spooler+1.0+-+Bambu.zip (37MB))
[Upload-project-file-(V-Spooler+1.0+-+Bambu.3mf)-is-not-possible.txt](https://github.com/user-attachments/files/17084408/Upload-project-file-.V-Spooler%2B1.0%2B-%2BBambu.3mf.-is-not-possible.txt)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Ringo-092024, 2024-09-21:
Best regards from Germany!

LiZ-Li-BBL, 2024-09-30:
We will fix it soon

ttt888zzz, 2024-10-16:
The latest version has been fixed.
 Please visit the website at https://github.com/bambulab/BambuStudio/releases， and try again.

closed 2024-10-16 by ttt888zzz