bambulab/BambuStudio #4865: Problems with new version 1.9.7.52

wagnerlip, 2024-09-21:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Lab A1

### How to reproduce

Please read the "Actual results"  below

### Actual results

Right after install this new version, I was unable to slice "OTHER", "BY OBJECT", even that Studio select the option, the slice shows slicing on both parts same layer, not individual parts.  Also, the separation gray area normally shown on plate under "object" disappeared, so, it is not working anymore, Used the BY OBJECT with a month older version this morning, not working now.

Second problem, the ARRANGE all objects on place also stop working, it is saying it cannot arrange because the plate is locked, it is not, I also exercise the lock/unlock several times, the arrange still saying the plate is locked.  

Something is odd with this version.

How can I install the month ago previous version????

### Expected results

Previous Studio version from a month ago was working nice.

### Project file & Debug log uploads

[Portao1.zip](https://github.com/user-attachments/files/17085265/Portao1.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

MartinChan3, 2024-09-22:
[image]
I believe what you want is like above picture.The plate setting was set to 'By layer' and you should change it to 'Same as Global Print Sequence'

wagnerlip, 2024-09-22:
Yes, thank you.
I found out this yesterday, but I never touched those plate icons
setup before.
Probably when installing the new version something happened that changed
that.
Thank you again.
Wagner Lipnharski

closed 2024-09-23 by fahaaaxiki