bambulab/BambuStudio #4867: Plate area in 'Prepare' and 'Preview' tabs not rendering.

TJKwispy, 2024-09-22:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab github releases

### OS version

Bazzite 40

### Additional system information

CPU: AMD Ryzen 9 5950X
GPU: NVIDIA GeForce RTX 3080 Ti

### Printer

Bambu P1P

### How to reproduce

Launch Bambu Studio
open Prepare/Preview Tab

### Actual results

View of the project does not render.

[image]

### Expected results

Screenshot taken from the Flatpak page for example.

[image]

### Project file & Debug log uploads

*No related project file.

[log.zip](https://github.com/user-attachments/files/17089071/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

MackBambu, 2024-09-23:
```got opengl version 4.6.0 NVIDIA 560.35.03, glsl version 4.60 NVIDIA, vendor NVIDIA Corporation```
same issue，fixes can be found here
https://github.com/bambulab/BambuStudio/issues/4669

MackBambu, 2024-11-18:
This is a known issue; try downgrading the driver to version 555.
Closing the issue, please discuss here：
https://github.com/bambulab/BambuStudio/issues/4669