bambulab/BambuStudio #4868: Move tool places object at random heights

owocowyy, 2024-09-22:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11 Pro

### Printer

P1S

### How to reproduce

1. Open 3MF file I attached
2. Try to place the "keyrambit" object flat with the plate (so they are touching each other) using the Move tool.
3. When you input 0 at the "Z" axis the object goes very deep beyond the plate and the actual "Z" becomes -23.44.
4. Now input 23.44 as the new "Z" and then the object goes very heigh (93.76 mm)

I encountered the same behaviour when using modifiers or just wanting to flip the object by 180 degrees (so I can get the texture from the build plate at the "front" of the model). When I flip the object and then want to place it flat with the bed using the move tool, "Z" axis starts behaving differently then before the flip operation.

### Actual results

The object appears at random heights

### Expected results

The object should be placed exactly at the height specified in the "Z" axis field

### Project file & Debug log uploads

[z_axis_bug.3mf.zip](https://github.com/user-attachments/files/17089812/z_axis_bug.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-09-23:
[image]
We have upgraded the translation tool in our internal version. Please wait for the next version. Thank you~

DevHyfes, 2024-09-24:
Hi Friends, I do not think your feature to select the coordinate system fixes this issue. Let me describe the behavior I am seeing. 

[image]

Notice that the current coordinates are X:83.59, Y130.18, Z17.66. I click in the "Z" field, and enter "0.00" and the object does not drop to the plate. Instead it shows this: 

[image]

The Z position has not changed to 0, it has doubled to 35.32

Maybe this upgrade fixes that problem, but I just wanted to be clear on the issue that I am having and that I think the Original Reporter was having.

DevHyfes, 2024-09-24:
playing around with this more, I can see that the dialogue box honors my selection for X value- if I enter 0, it will move the piece to 0, and the field will update to 0.00. But if I try the same with Y and Z, I get the same doubling behavior.

Haidiye00, 2024-09-25:
[image]
this is my test result

DevHyfes, 2024-09-25:
Hi @Haidiye00 - In case it matters, the item that was not behaving correctly is a MODIFIER for the main part. It is also an SVG that I imported and flattened. Otherwise, I don't really know what to tell you. I just loaded up again, imported the file, and tried moving it by entering 0 into the Z field and hitting enter...it goes to a random space. TODAY the behavior is, if I enter 0 into the Y field, the Z field doubles, and if I enter 0 into the Z field, the Y field doubles. I am able to work around it by Centering the modifier but I cannot do precision placements, which is annoying.

Haidiye00, 2024-09-26:
Sorry，at present, there should exist a bug in your version, which can be resolved by manually dragging the arrow. Please wait for the new version to upgrade this feature.

DevHyfes, 2024-09-27:
Thanks for the update!

Haidiye00, 2024-10-16:
The new version 1.10beta has been released. Please download and use~

closed 2024-10-16 by Haidiye00