bambulab/BambuStudio #4870: Multi color fill 2nd to top layer becomes the color of the body

Coop25, 2024-09-22:
### Bambu Studio Version

1.09.07.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

CPU: AMD Ryzen 9 3900X
Memory: 64 GB DDR4
Video Card: RTX 4070 Ti

### Printer

bambu lab X1C, P1S

### How to reproduce

import model with text sunk in to it and try to fill the text with a different color from the original model

### Actual results

I get the top 3 layers of the fill to look like the gif below

[image]

### Expected results

all 3 of the layers above should be the same color and are not

### Project file & Debug log uploads

[debug_Sun_Sep_22_10_52_42_25012.log.zip](https://github.com/user-attachments/files/17090297/debug_Sun_Sep_22_10_52_42_25012.log.zip)
[project file.zip](https://github.com/user-attachments/files/17090303/FAFO.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

zhimin-zeng-bambulab, 2024-09-30:
It will be fixed in the next version. Thanks for reporting this issue.

ttt888zzz, 2024-10-16:
The latest version has been fixed.
 Please visit the website  at https://github.com/bambulab/BambuStudio/releases and try again.

closed 2024-10-16 by ttt888zzz