bambulab/BambuStudio #4871: Color Painting Tool - Cannot Unselect Height Range

Wingeddonkey, 2024-09-22:
### Bambu Studio Version

1.9.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Pro, 22H2

### Printer

Bamvu Carbon X1

### How to reproduce

Press N
Ensure dropdown menu overlaps printed object
Select Height Range
Select Fill tool (Unable)

### Actual results

When the dropdown menu for paint, overlaps the object to be painted, the menu is not treated as the highest clickable action. Rather the height tool remains priority. 
Unless you move the object around so the menu doesnt overlap it, you cannot unselect the heaight tool. 

### Expected results

Ability to select new tool option or paint color.

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-09-23:
We have identified and fixed this issue in our internal version. Thank you for your discovery~

ttt888zzz, 2024-10-16:
The latest version has been fixed.
 Please visit the website  at https://github.com/bambulab/BambuStudio/releases and try again.

closed 2024-10-16 by ttt888zzz