bambulab/BambuStudio #4877: AppImage missing dependencies

mihalyr, 2024-09-23:
### Bambu Studio Version

v01.09.07.52

### Where is the application from?

Bambu Lab github releases

### OS version

Fedora 40 Sericea

### Additional system information

N/A

### Printer

N/A

### How to reproduce

1. Download the AppImage
2. Try to run the AppImage

### Actual results

```
➜  Downloads ./Bambu_Studio_linux_fedora-v01.09.07.52.AppImage 
/tmp/.mount_Bambu_RfNPJu/bin/bambu-studio: error while loading shared libraries: libsoup-2.4.so.1: cannot open shared object file: No such file or directory
```

### Expected results

it runs

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [X] Log file
- [X] Project file

mihalyr, 2024-09-23:
Once I installed the dep, another dep is missing

```
(fedora-toolbox-40) ➜  Downloads ./Bambu_Studio_linux_fedora-v01.09.07.52.AppImage 
fuse: failed to exec fusermount: No such file or directory

Cannot mount AppImage, please check your FUSE setup.
You might still be able to extract the contents of this AppImage 
if you run it with the --appimage-extract option. 
See https://github.com/AppImage/AppImageKit/wiki/FUSE 
for more information
open dir error: No such file or directory
```

mihalyr, 2024-09-23:
After installing `libsoup` and `fuse` now another library is missing...

```
(fedora-toolbox-40) ➜  Downloads ./Bambu_Studio_linux_fedora-v01.09.07.52.AppImage 
/tmp/.mount_Bambu_l6kJJL/bin/bambu-studio: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory
```

mihalyr, 2024-09-23:
After installing `libsoup`, `fuse`, `webkit2gtk4.0` now another dependency is missing

```
(fedora-toolbox-40) ➜  Downloads ./Bambu_Studio_linux_fedora-v01.09.07.52.AppImage 
/tmp/.mount_Bambu_9ShhGq/bin/bambu-studio: error while loading shared libraries: libOSMesa.so.8: cannot open shared object file: No such file or directory
```

mihalyr, 2024-09-23:
Finally it worked, so on Fedora 40 at least. This is the list of dependencies that were missing on Fedora 40 from the `Bambu_Studio_linux_fedora-v01.09.07.52.AppImage`

* `libsoup`
* `fuse`
* `webkit2gtk4.0`
* `mesa-libOSMesa`

mihalyr, 2024-09-23:
Hm, I tried OrcaSlicer (which BambuStudio seems to be some kind of fork of), but it has the same problem, as soon as I click Preview or Slice, it will fail with a segmentation fault. 

I installed Cura, that works, but couldn't find my printer for some reason.

Then I came across a Flatpak version of BambuStudio that just works out of box! 
See this: https://flathub.org/apps/com.bambulab.BambuStudio

I would recommend releasing this officially instead of the broken AppImage.

MackBambu, 2024-09-27:
appimage is not available out of the box on all Linux systems yet, and the flatpak version is a good alternative. But it is released by a third party.

ExperimentalCyborg, 2024-10-01:
+1, experiencing the same issues as above.

hadess, 2024-10-09:
> appimage is not available out of the box on all Linux systems yet, and the flatpak version is a good alternative. But it is released by a third party.

You're listed as one of the maintainers:
https://github.com/flathub/flathub/issues/5220

Either you or @lanewei120 can do what's needed to get the app verified.

In the meanwhile, it's true that I'm the one doing all the work on that repository, with no help or compensation from Bambu Lab.

lanewei120, 2024-10-10:
For these system AppImage can not be used, we also suggest to use the flathub version from https://flathub.org/apps/com.bambulab.BambuStudio

as we mentioned before, it is totally developed and maintained by @hadess 
thanks again for @hadess 's hard work

mihalyr, 2024-10-10:
> we also add flathub links into the BambuStudio's Readme

Thank you, the flatpak works great!

mihalyr, 2024-10-10:
I think I missed the flatpak link on the README, because the website Linux download takes me directly to the releases page on GitHub where there is no flatpak nor flathub link, so it's easy to miss.

tr0043t, 2024-11-03:
I'm in a similar situation. I could never get the AppImage versions of Studio to run on Ubuntu. The flatpak -- great work @hadess ! -- runs great, with one odd exception. I've never been able to get left-click drag movement of objects to work. No matter what I try, the object is selected but there is no movement when I left-click drag. I can move objects via various other methods. Keyboard arrows work. The Move tool works and I can drag along each axis individually with the tool arrows. But I have yet to find a way to freely move an object with the mouse. Maybe I'm missing something, but I've spent considerable time on this and never found a solution.

ExperimentalCyborg, 2024-11-03:
What works on Fedora 40 for me:
Get the ubuntu2404 .AppImage of OrcaSlicer

Make sure these are installed:
- libsoup
- fuse
- webkit2gtk4.0
- mesa-libOSMesa

Set these environment variables:
`__GLX_VENDOR_LIBRARY_NAME=mesa`
`__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json`

It does almost everything it should, though for me it freezes when i send a print to a P1S that is in LAN mode. I just walk SD cards back and forth. Other than that everything seems to work fine, including AMS filament detection with RFID spools :)

tr0043t, 2024-11-03:
I really want to stick to one slicer -- Studio -- at this point. I've been through the multiple slicers bit too many times in the past! The Studio AppImages for me get a pile of GLIB errors. Apparently a bunch of the dependencies are not built in.

ExperimentalCyborg, 2024-11-03:
I get those errors too. They don't seem to break anything i need to produce gcode files.

Maybe this is a good solution for people like me, who couldn't get bambu slicer to work at all.

tr0043t, 2024-11-03:
In my case I get the errors and the program immediately exits.

ExperimentalCyborg, 2024-11-03:
There is most likely another dependency error hiding behind the GLIB spam! I had that too until i installed the dependencies i listed above. Maybe something is missing from that list, since i have a bunch of other stuff installed that could have brought more libs in that orca coincidentally might use.

tr0043t, 2024-11-03:
I've been through the lists pretty carefully. There are some dependencies that do not have installable versions available for the target machine. Really, the flatpak version is nifty and just works, except for that one issue I noted.

PatrickGerth, 2026-10-29:
The same bug has been plagueing the appimage for ages now. Would it not be possible to update the dependencies within the sourcecode, so that modern environments can finally use the regular appimage again? I mean I am grateful for the flatpak but on the latest Fedora some buttons seem misaligned and many controll-elements are not shown in the correct way. Please send someone to update the original sourcecode. After more than 4 months this has become embarrassing for a company of your size.