bambulab/BambuStudio #4878: Can not "send" file over Internet to my A1 mini, but I can "print" the same file directly over cloud...

ckupcu, 2024-09-23:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Pro (22H2)

### Additional system information

Intel(R) Core(TM) i7-6770HQ CPU @ 2.60GHz   2.59 GHz
32.0 GB (31.9 GB usable)
OS Build: 19045.4780

### Printer

BambuLab A1 Mini

### How to reproduce

1. Just download any 3D model from BambuLab OnLine Models
    - I downloaded: Compact poop bucket S (A1, A1 mini).3mf
2. Position the model in Prepare screen
3. Slice it in the Preview screen
4. Try to send it to a printer on another network (ie in the office), where I use BambuLab SW at home
[image]

### Actual results

As in the attached picture, BambuLab SW can not connect my remote A1 printer to "Send" the file. It asks me to be on the same LAN for some reason...

### Expected results

... where as if I try "Print" command, very same sliced file can be transferred to the remote printer via BambuLab cloud, file get stored on printer's SD card, and print start. If I stop the print, I can see the transferred print file in the SD card!

1. Does the "Send" command requires using VPN to access remote printer?
2. Is relaying/sending the file over BambuLab cloud without printing not possible?

Rgds,
Cengiz

### Project file & Debug log uploads

Not needed, can be tried with any file...

### Checklist of files to include

- [ ] Log file
- [X] Project file

ckupcu, 2024-09-23:
I can download for instance "Timelapse" video to my PC from the remote A1 Mini printer's SD card in the office - this works.

I expects, sending (uploading) a file in the same way should also work.

walterwongbbl, 2024-10-08:
@ckupcu 
We will soon support remote send, please look forward to the next version.

ckupcu, 2024-11-07:
> @ckupcu We have released 1.10 Beta2 to solve the similar problem, download https://github.com/bambulab/BambuStudio/releases/tag/v01.10.00.81 to enjoy subnet connection to your A1 mini.

Hi Fahaaakiki,
I installed and tried BambuStudio 1.10 Beta2 on my MacBook:- I can send files when I am remotely connected over VPN - OK- I cannot send files remotely over Bambu cloud or directly, I still get this message:

Hope this helps...
Rgds,Cengiz=====
Cengiz Tugsav KÜPÇÜ
Zürichstrasse 77F, CH-8134 Adliswil/Switzerland
Tel: +41 44 480 27 49, Mobile: +41 79 733 32 72
e-mail: ***@***.*** / ***@***.***

-----------------------------------------------------------------------------------------------I don't want to be human! I want to see gamma rays, I want to hear X-rays, and I want to smell dark matter ... 
I want to reach out with something other than these prehensile paws and feel the solar wind of a supernova flowing over me.

GobiTodic, 2025-05-27:
That doesn't work for me either. Is this a bug or a missing feature?

fahaaaxiki, 2025-05-28:
@ckupcu @GobiTodic Sorry, currently, A1 mini does not support sending via the cloud. We will support the H2D  printer and the X series in the next version, and the A series is still under planning.