bambulab/BambuStudio #4879: Supports grams  (A LOT ) are incorret with multi buildplates

Duudeeee, 2024-09-23:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10

### Additional system information

intel7

### Printer

X1C

### How to reproduce

just add suuports

### Actual results

i have a few buildplates with stuff ready to print with minimal supports. and when i slice it give me the double grams, without supports 600g , with 1200g. its impossibel someting its wrong here

### Expected results

to much grams

### Project file & Debug log uploads

[image]
[10cm.zip](https://github.com/user-attachments/files/17105125/10cm.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

LiZ-Li-BBL, 2024-09-24:
Thanks for reporting this issue. We're fixing it.

ttt888zzz, 2024-10-16:
The latest version has been fixed.
Please visit the website at https://github.com/bambulab/BambuStudio/releases and try again.

closed 2024-10-16 by ttt888zzz