bambulab/BambuStudio #4881: 支撑统计不准确

TinCivil, 2024-09-24:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Win10

### Additional system information

CPU: Intel i5-12400
内存:16GB
GPU: Intel Graphics 730

### Printer

P1S

### How to reproduce

1.点击预览
2.点击所有打印板统计
3.即可发现支撑统计的重量和耗材长度不准确

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

[Uploading 1.zip…]()

### Checklist of files to include

- [X] Log file
- [X] Project file

LiZ-Li-BBL, 2024-09-24:
我们已经修复此问题，请等待下个版本

LiZ-Li-BBL, 2024-09-24:
#4879

ttt888zzz, 2024-10-16:
The latest version has been fixed.
Please visit the website at https://github.com/bambulab/BambuStudio/releases and try again.

closed 2024-10-16 by ttt888zzz