bambulab/BambuStudio #4882: Missbehavior when cutting parts

Sectorchan, 2024-09-24:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Win 10

### Additional system information

Ryzen 7 5800x3d
32GB RAM
RTX 4080

### Printer

A1

### How to reproduce

[image]
This object shall be cutted 1mm above the plate (Since multiple object can not be cutted at a single time)
[image]
I dont need the blue part. Then I press Perform cut.
[image]
Now the Part is below the other object, this shoulnt happpen.

### Actual results

See above

### Expected results

I expect that the part is not moved.
OrcaSlicer behaves like that:
[image]
[image]
[image]

### Project file & Debug log uploads

[object.zip](https://github.com/user-attachments/files/17113184/object.zip)
[debug_Mon_Sep_23_15_52_16_15100.log.zip](https://github.com/user-attachments/files/17113224/debug_Mon_Sep_23_15_52_16_15100.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Haidiye00, 2024-09-25:
Hello, could you please record a GIF? Thank you? Facilitate us to reproduce the problem and solve it

Sectorchan, 2024-09-26:
[image]
This is from Orcaslicer

[image]
This is from Bambu

Haidiye00, 2024-09-26:
can i get your 3mf?

Haidiye00, 2024-09-26:
I understand, thank you

Sectorchan, 2024-09-26:
Glad I could help!

Haidiye00, 2024-09-27:
[image]
Thank you, we will make improvements. At present, you can hook the cut to parts and then delete the parts you don't want without changing their position.

Haidiye00, 2024-10-16:
The new version 1.10beta has been released. Please download and use~

closed 2024-10-16 by Haidiye00