bambulab/BambuStudio #4886: Adding Custom Filament

ThisMustbeFabricating, 2024-09-24:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS14.6.1

### Additional system information

Macbook Pro M3

### Printer

Bambu X1 Carbon

### How to reproduce

Follow the steps detailed on the Bambu Wiki for creating custom filament, method one.  Attempt to select the new custom filament directly on the Bambu X1C touchscreen. Repeat.

### Actual results

After creating a new custom filament and choosing to Copy from an existing filament, I attempt to select the new Custom Filament on the Bambu X1C touchscreen and after 3-5 seconds, the selection deselects and the filament slot returns to white color with a ? in the middle.  Sometimes it will also change back to the settings I had set, only to revert once more after another 3-5 seconds.  

### Expected results

To be able to select my custom filament types and profiles from the touchscreen on the printer when I load filament in, so that it will then synchronize with Bambu Studio.  This seems like a basic feature and there are many external posts about it and how following the exact instructions from the Bambu Wiki (https://wiki.bambulab.com/en/bambu-studio/create-filament) works intermittently at best and not at all at worst. 

### Project file & Debug log uploads

I'm neglecting to include a project file since this is not project-specific. This regards functionality of the printer firmware and Bambu Studio.
[BambuLogs.zip](https://github.com/user-attachments/files/17122065/BambuLogs.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

psjshep, 2024-10-07:
I am experiencing the exact same issue:
Create filament -> custom vendor=Geeetech, Type=ABS, Serial=Basic

After selecting filament on devices screen, after a few seconds it reverts back to ?

DanBao-Bambu, 2024-10-12:
From the log, it appears that you have set slot information multiple times on Studio. But I didn't see any information about Studio resetting the slot independently. Please first check if there are multiple Studio instances running (including on different hosts). If not available, a printer log will be required. Please export and send it to me. Thank you.

ThisMustbeFabricating, 2024-10-16:
Ah thank you. I wasn't aware multiple instances may cause these problems
and indefinitely have a habit of keeping my next couple projects open for
slicing while printing another.

I will retry adding custom filament with only a single instance open. Thank
you for your help!