bambulab/BambuStudio #4891: Flow dynamic calibration with AMS not working as expected

daibaron, 2024-09-25:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 pro

### Printer

X1C + AMS

### How to reproduce

1- Go to Calibration
2 - Flow Dynamic
3 - Check all boxes and select filament presets "
4 - run calibration "Calibrate'

### Actual results

Take in consideration that during a regular print everything works well (color switching, purge) the overall print quality is OK

But during calibration with AMS I'm reporting the following issues.

When I want to calibrate 4 filaments from AMS in A1 A2 A3 A4 slots
Filament from A2 slot is not purging long enough and calibration is mixed up with the previous A1 slot filament remains (a weird sound from AMS occurs) colors are accurate at the beginning
A1 A3 A4 filaments calibration result is OK not A2

Same issue if I want to calibrate 3 filaments from AMS in A1 A3 A4 slots
- Filament from A3 slot is not purging long enough and calibration is mixed up with the previous A1 slot filament remains (a weird sound from AMS occurs) colors are accurate at the beginning 
A1 A4 filaments calibration result is OK not A3

so you have to understand the 2nd filament is not purged correctly in every cases

### Expected results

All filaments are calibrated with accurate colors

### Project file & Debug log uploads

[debug_Wed_Sep_25_12_16_57_20784.log.zip](https://github.com/user-attachments/files/17129491/debug_Wed_Sep_25_12_16_57_20784.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

daibaron, 2024-09-26:
https://github.com/bambulab/BambuStudio/issues/4650

Qjeezy, 2024-09-29:
Babu, please just fix this already. It can't be that hard to adjust the flush volume from the first to the second filament.

gaolegao-lx, 2024-10-10:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

be221b, 2024-10-18:
> **Please upload log through Handy APP, and give us your ticket number**
>
> _**Upload log files through Bambu Handy**_ Log files from the Bambu printer can be uploaded via Bambu Handy over the network, [...]

What is this response supposed to accomplish? This is an issue with the flow dynamic calibration? Is the issue not universal/repeatable?

daibaron, 2024-10-18:
Not able to do this I'm far from home for a while