bambulab/BambuStudio #4897: "Arrange all objects" incorrectly complains about locked objects

rgov, 2024-09-25:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.1

### Printer

Bambu Lab P1S

### How to reproduce

1. Under the "Prepare" tab, set up two filaments.
2. Add three cube primitives to the plate.
3. Select two of the cubes and right-click and merge them into an assembly.
4. In the sidebar, set your standalone cube to one filament and your assembly to the other filament.
5. Change "Print sequence" to "By object."

    You should receive the warning "Assembly is too close to others, and collisions may be caused."

6. Click "Arrange all objects" in the toolbar.

    You should receive the warning "All the selected objects are on the locked plate, We can not do auto-arrange on these objects."

### Actual results

The auto-arrange toolbar item does not work because it incorrectly claims the objects are locked. 

### Expected results

Auto-arrange works fine.

### Project file & Debug log uploads

x

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tangketan, 2024-09-27:
It's not a bug. Your plate's printing sequence differs from the global setting, so the global arranging command ignores this plate.

rgov, 2024-09-27:
Thanks for reviewing the bug. The behavior seems very counterintuitive to me.  Why should modifying a setting (is it *only* printing sequence? or any plate setting?) disable auto-arrange, when the plate has its own auto-arrange? Why does the message say something about being locked (the icon is very clearly *unlocked*)? How is the user supposed to know which toolbar buttons work or don't work when settings have been changed?