bambulab/BambuStudio #4898: Random Access violation when slicing

TennisRunner, 2024-09-26:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

p1p

### How to reproduce

Load model, slice.
Maybe change the support top z distance to 0.0 and 0.2.
Its inconsistent. 

### Actual results

Access violation crash. Sometimes crashes with no error message at all. 

[capture-9_25_2024 7-52 PM.webm](https://github.com/user-attachments/assets/0a6128e0-6bde-4a60-9008-3dff119b3715)

### Expected results

Shouldn't crash.

### Project file & Debug log uploads

[rounded-mid-floater.zip](https://github.com/user-attachments/files/17140064/rounded-mid-floater.zip)
Not including dump file since it contains too much personal info.

### Checklist of files to include

- [X] Log file
- [X] Project file

DRA6N, 2024-09-26:
I had the same issue happening. Had to roll back to previous version, ultimately switched to Orca and never looked back.

QingZhangBambu, 2024-10-11:
@TennisRunner it had been fixed, and will released in next version, please waiting
[image]

ttt888zzz, 2024-10-16:
The latest version has been fixed.
 Please visit the website  at https://github.com/bambulab/BambuStudio/releases and try again.

closed 2024-10-16 by ttt888zzz