bambulab/BambuStudio #4900: Support for ABS will recognize as ABS

QuokeCola, 2024-09-26:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Windows 10

### Printer

Bambu X1C

### How to reproduce

1. Load bambu support for ABS in AMS.
2. Go to support and enable support.
3. When choosing filament for supports, the material selection for bambu support for ABS turns to ABS.
[image]

### Actual results

This file is printable from bambu studio since it treat the ABS and support for ABS as one material.
[image]

But it cannot be printed again through the x1c screen, since AMS is able to correctly recognize the support for ABS filament, and X1C treat ABS and support for ABS as two different material.

 

### Expected results

Bambulab is able to treat the support for PLA correctly. In material selection interface, the filament is treated as Sup.PLA

[image]

Support for ABS should have similar behavior, in bambu studio it should be treated correctly as Sup.ABS but not ABS.

### Project file & Debug log uploads

[astranaut.zip](https://github.com/user-attachments/files/17143021/astranaut.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

XunZhangBambu, 2024-09-29:
@QuokeCola  Thanks for the feedback. Will fix the issue soon

ttt888zzz, 2024-10-16:
The latest version has been fixed.
 Please visit the website  at https://github.com/bambulab/BambuStudio/releases and try again.

closed 2024-10-16 by ttt888zzz

lumbduck, 2025-05-02:
I still have this issue on the latest release in May 2025. Support for ABS is not accessible from the X1C screen, where it believes that ABS has been designated. Workaround is to use 3rd party support material so that the AMS doesn't recognize it, but I would love to have this working correctly