bambulab/BambuStudio #4901: Sagfault when auto arranging

Tnthr, 2024-09-26:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab github releases

### OS version

Linux Mint 21.3

### Printer

X1C

### How to reproduce

1. Open the model https://makerworld.com/en/models/434740
2. change the text item "Type" to say "SILK"
3. clone the model 3 times
4. auto arrange the plate

### Actual results

segfault and the program crashes
[Sep26 09:08] bambustu_main[879788]: segfault at 5d1f0492657e ip 00005d1ad0e2f2a8 sp 00007ffcb3c8d628 error 4 in bambu-studio[5d1acfb8e000+3aa5000] likely on CPU 16 (core 32, socket 0)
[  +0,000021] Code: 00 90 b8 01 00 00 00 83 fa 02 74 06 c3 0f 1f 44 00 00 48 8b 06 48 83 b8 b0 00 00 00 00 0f 95 c0 c3 90 48 8b 06 48 85 c0 74 12 <83> b8 98 00 00 00 01 74 07 48 8b 00 c3 0f 1f 00 31 c0 c3 90 0f 1f

### Expected results

the plate will auto arrange

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/17144642/log.zip)
[project.zip](https://github.com/user-attachments/files/17144645/project.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

tangketan, 2024-10-10:
The issue can't be reproduced for now. Will keep tracking.

cyberang3l, 2025-06-23:
Experiencing the same issue on v2.1.1.52, running BambuStudio from the ubuntu AppImage (Bambu_Studio_ubuntu-24.04_PR-7292.AppImage)