bambulab/BambuStudio #4903: New Bambu Studio version crashing when slicing previously okay plate

PaviFlye, 2024-09-26:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 22H2 19045.4894

### Additional system information

14th Gen i7-14700k, 32gb ram, 2tb WD m.2, GeForce RTX 4070ti Super

### Printer

Bambu Lab X1

### How to reproduce

Attempt to slice "Board_X1", Generic PLA Red

### Actual results

Bambu Studio Crashes.

### Expected results

It slices the plate for printing.

### Project file & Debug log uploads

[Connect4_Vortext_Full.zip](https://github.com/user-attachments/files/17152088/Connect4_Vortext_Full.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Jvigeant1969, 2024-10-19:
I am having the same issue as above and tried the steps in the link but still super unstable. 

Is there a place where i can download an older version the bambu Studio? 

i never had any issues until recently . 

My machine should be powerful enough: 

Win11 Latest updates
14th gen i9-14900K
128 Gb  ram
GTX 4090

ttt888zzz, 2024-10-21:
HI [Jvigeant1969](https://github.com/Jvigeant1969) :  
If you want to  download the old version website，please visit https://github.com/bambulab/BambuStudio/releases
[image]