bambulab/BambuStudio #4904: Unnecessary isolated thin bridges that break and create plastic debris inside hollow models

tokamac, 2024-09-26:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab github releases

### OS version

macOS Sequoia 16.0

### Additional system information

Apple Silicon M1

### Printer

Bambu Lab X1-Carbon

### How to reproduce

This bug has been introduced in Bambu Studio 1.9.0 Public Beta 2 (v01.09.00.69) and remains in all subsequent versions since, including latest current version 1.9.7.52 Public Official. 
This bug didn't exist in 1.9.0 Public Beta (v01.09.00.60) or any previous versions of Bambu Studio. 

Steps to reproduce:

1. Open the attached 3MF file in Bambu Studio 1.90 PB2 or any newer version. This 3MF file contains a portion of a larger 3D model that was cut to show only the area where the problem arises.
2. Hollow the model using 0% infill
3. Slice
4. Parse the sliced model in the Preview pane using the vertical green bar, paying attention to the blue bridges.
5. Redo the same thing in Bambu Studio 1.9.0 PB or any prior version for comparison.

### Actual results

[image]

You'll notice that in latest versions of Bambu Studio from 1.9.0 Public Beta 2 onwards, just under the angled part inside the object, two unnecessary flat thin bridges are formed, that run right through the object's interior. They seem to support the birth of a tiny protrusion from the wall on about 1 mm, but 99% of these bridges supports nothing and just hang as is in thin air in the middle of the hollow object. Being very thin and isolated from anything else except their two ends connected on the walls, they are very fragile. As a consequence, they end up breaking inside the full, closed object printed, and the debris rattles around inside, unable to get out. 

This is so unbearable for my application that I am forced to locate the lines of the Gcode that produces these two bridges by parsing the horizontal green bar in the preview pane of the sliced model, export the Gcode, open it in a text editor, locate these lines, erase them manually, save and overwrite the file, copy it to the SD, go to the X1C to insert the card, and print from there. I have to repeat this mad cumbersome process each time I update Bambu Studio, of for any iteration of the model, which has various sizes and orientations.

This exact same model printed just fine without these breaking unnecessary isolated thin bridges before.

### Expected results

No unnecessary isolated bridges, as that was the case before for that model, since the beginning of Bambu Studio up to 1.9.0 Public Beta. 

Please make my Bambu workflow a pleasure again, instead of the current nightmare!

### Project file & Debug log uploads

[Bug Bridges.zip](https://github.com/user-attachments/files/17152558/Bug.Bridges.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

tokamac, 2024-09-26:
Reading the [release notes from Public Beta 2](https://github.com/bambulab/BambuStudio/releases/tag/v01.09.00.69), my issue may be related to the new "feature" described at point 7:

> **7. Port the "Extend sparse infill" feature from PrusaSlicer. Thanks to @Godrak** 
> 
> [image] 
> [image]

Although at the moment I don't get how this can be a positive addition.

XunZhangBambu, 2024-09-29:
@tokamac  Sorry for the inconvenience. In your model, the bridge was generated to support the solid infill. And the solid infill was generated to essure the vertical thickness. To remoev the bridge,you can add a height range modifier and set the wall count to 4 from 2.4mm~5mm
[image]

[image]

XunZhangBambu, 2024-10-11:
@tokamac  We are considering support fully disable "ensure vertical shell thickness" . Currently, disabling this option will use another algorithm to generate such infill.