bambulab/BambuStudio #4906: Total suppport filament is wrong

zmindless, 2024-09-27:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu X1C

### How to reproduce

not sure, try using 12 plates and slice all, used 3 materials

### Actual results

[image]
support total is just copy/paste from model total

### Expected results

support total is actual total of all support listed above

### Project file & Debug log uploads

[Issue.zip](https://github.com/user-attachments/files/17163121/Issue.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

LiZ-Li-BBL, 2024-09-29:
Same as https://github.com/bambulab/BambuStudio/issues/4879

ttt888zzz, 2024-10-16:
The latest version has been fixed.
Please visit the website at https://github.com/bambulab/BambuStudio/releases and try again.

closed 2024-10-16 by ttt888zzz