bambulab/BambuStudio #4909: Crashes after send to printer

sbeerda, 2024-09-27:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.0 (24A335)

### Additional system information

Mac mini M2 8GB

### Printer

P1S

### How to reproduce

slice, print plate, send, upload file and then crashes

### Actual results

studio crashes

### Expected results

should of to device screen

### Project file & Debug log uploads

does with all projects

### Checklist of files to include

- [ ] Log file
- [ ] Project file

skylerwshaw, 2024-10-13:
I am also experiencing this crash on version `1.9.7.52` with an X1C printer. MacOS 14.7. 2021 Apple MBP M1 Pro.

skylerwshaw, 2024-10-13:
This appears to be a dupe of https://github.com/bambulab/BambuStudio/issues/4857

walterwongbbl, 2024-10-24:
@skylerwshaw 
This is due to compatibility issues with the UI components used by macOS.