bambulab/BambuStudio #4911: 材料统计bug

clan4456, 2024-09-28:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab github releases

### OS version

win11

### Printer

P1SC

### How to reproduce

1、切片所有盘
2、统计材料结果出错，支撑用料总计出错。
[image]

### Actual results

[image]

### Expected results

支撑用料总计应为99.83g

### Project file & Debug log uploads

[可达鸭组装图.zip](https://github.com/user-attachments/files/17173759/default.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

LiZ-Li-BBL, 2024-09-29:
已修复此问题 https://github.com/bambulab/BambuStudio/issues/4879

ttt888zzz, 2024-10-16:
The latest version has been fixed.
Please visit the website at https://github.com/bambulab/BambuStudio/releases and try again.

closed 2024-10-16 by ttt888zzz