bambulab/BambuStudio #4914: Can't fully load 3mf files after they have been saved in Orca Slicer. Prompts to update Bambu Studio.

Chewieez, 2024-09-28:
### Bambu Studio Version

1.9.7.5.2

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Sonoma 15.0

### Additional system information

Macbook Pro 14" M3 

### Printer

Bambu Lab P1S

### How to reproduce

1. Open .3mf file in Orca Slicer and save it.
2. Open the same file in Bambu Studio
3. See if you see an error popup that says "Due to the lower version of Bambu Studio, this 3mf file cannot be fully loaded. Please update Bambu Studio to the latest version"

### Actual results

I am using Orca Slicer 2.1.1.

I have tried fully erasing Orca Slicer (arm) and Bambu Studio, including erasing their files in User/Library/ApplicationSupport. 
I've tried 2 beta released of Orca Slicer. I've tried using the x86 version of Orca as well. None of these fixed the issue.

[image]

### Expected results

Opening the 3mf fully and not losing any available settings. 

### Project file & Debug log uploads

[Archive.zip](https://github.com/user-attachments/files/17176643/Archive.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Chewieez, 2024-09-29:
Maybe I just haven't noticed it before, or I've been lucky. I've been using both Bambu Studio and Orca Studio since Nov and just noticed this happening in the past few weeks. 

I also realize I wasn't as clear as I could have been in the steps to reproduce. 

In my case, I am downloading a 3mf from Makerworld, making some changes in Orca, then attempting to open the file in Bambu Studio and I get that error. But again, maybe I just didn't notice that this never worked. :)

ttt888zzz, 2024-09-30:
HI [Chewieez](https://github.com/Chewieez)：It's not a bug.BambuStudio accepted the 3mf standard, while Orca did not, which led to the error.

Achimoto, 2025-01-06:
You can always strip the settings of any slicer with a simple batch file (Windows, using 7Zip):

Create a .bat file, e.g. "Strip3mf.bat" with the content "C:\Program Files\7-Zip\7z.exe" d %1 "Metadata"
Drop .3mf files to the batch icon.