bambulab/BambuStudio #4918: Weird split only happens to some layer although every layer is the same

ccl13, 2024-09-30:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Windows 10

### Printer

Bambu Lab X1C

### How to reproduce

Please find attached model file. When using the parameters I have, it will process the model as split only for layer 9 to 50; although the model is the same on all layers.
The reason might be that I have a model that is of multiple parts. But the behavior is not correct.

### Actual results

It will process the model as split only for layer 9 to 50; although the model is the same on all layers. Please find screenshots.
[image]
[image]

### Expected results

The expected behavior is that either:
1) It processes all layers as split parts, since all the layers are the same.
2) Or it processes all the layers as combined model, again since all the layers are the same.

### Project file & Debug log uploads

[3M Command Strip Wall Cable clip.issue.zip](https://github.com/user-attachments/files/17184200/3M.Command.Strip.Wall.Cable.clip.issue.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

ccl13, 2024-09-30:
Can confirm that this happens with all printer setting, filament setting, and process setting that I have.

MechInnovation, 2024-10-04:
Just a suggestion, but wouldn't it be best to create a single model anyway. At the moment it is 2 seperate STL files that but up against oneanother. If you want it as one piece, just model it as one piece to avoid the issue altogether.

ccl13, 2024-10-04:
> Just a suggestion, but wouldn't it be best to create a single model anyway. At the moment it is 2 seperate STL files that but up against oneanother. If you want it as one piece, just model it as one piece to avoid the issue altogether.

I totally agree that's the better practice and I actually do that.
This bug is about inconsistent behavior - I could imagine that for people accidentally put a two part model that is closely touching each other, for example downloaded something from Internet without noticing, will be puzzled that why there's a dent on their print only on some layers. The expected outcome is that either all layers are processed as a single model, or all layers are processed as separate model, not that part it follows a different rule.

ttt888zzz, 2024-10-08:
HI [ccl13](https://github.com/ccl13)：
Please try following the steps below，and click the “slice plate”button
[image]

ccl13, 2024-10-08:
> HI [ccl13](https://github.com/ccl13)： Please try following the steps below，then slice again ...

Thanks for getting back to me! I've tried to fix the model, but somehow it got worse:

[image]

ttt888zzz, 2024-10-08:
HI [ccl13](https://github.com/ccl13)：
Please try following the steps below，then slice again
[image]

ccl13, 2024-10-08:
> HI [ccl13](https://github.com/ccl13)： Please try following the steps below，then slice again ...

Thanks. Merge the parts works as expected and address this issue. Either union or a simple merge works. However it is not the point of this bug report.

QingZhangBambu, 2024-10-09:
@ccl13 the corner of the part is simplified, so the result is wrong. you can adjust the resolution setting to get the right result. 
[image]
[image]
[image]

ccl13, 2024-10-10:
> @ccl13 the corner of the part is simplified, so the result is wrong. you can adjust the resolution setting to get the right result. ...

Yes this fixed the issue. But I am still confused as well as to the point of this bug report, why the resolution is inconsistent on different layers? If the resolution is simplify the model, I would think it should do the same to all the layer with the same situation. Because I don't think a 0.0 resolution is a viable value for all slicing profiles.

QingZhangBambu, 2024-10-11:
we check the model mesh. These two models do not completely stick together. it will have gap in the middle