bambulab/BambuStudio #4919: Live View: Player is Malfunctioning. Please reinstall the system player [2:2]

Mr-121, 2024-09-30:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.0

### Additional system information

Intel CPU

### Printer

X1C

### How to reproduce

Just updated the Bambu Lab Studio

### Actual results

Live view on Studio not working but works on Handy.

### Expected results

Live view should still work on Mac.

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

jebabi, 2024-09-30:
This is because you havn't allowed Bambu Studio access to your local network.
After updating to MacOS 15.0 it probably asked you the first time launching Bambu Studio. 

You can fix this by:
System Setting -> Privacy & Security -> Local Network

Mr-121, 2024-10-01:
You are right! Thanks!

Mr-121, 2024-10-01:
Maybe you guys can improve the error code to be more clear in next update. Thank you!

closed 2024-10-01 by Mr-121

peteramelang, 2024-11-10:
Did not help. I have also deleted everything from Application Support and reinstalled BambuStudio and the network driver.

I can watch the video feed or access the SD card from my Mac.
No issues with access through the mobile app.

IOIO72, 2024-12-16:
It doesn't work on one of my Macs as well.
Both have version 1.10.1.50 installed.
Both are running macOS Sequoia 15.2.
On both machines BambuLab has access to the local network.

It runs on my M1 Mac Mini.
It doesn't run on my M2 MacBook Pro 2023.

3driffs, 2024-12-16:
Ditto here, and it is enabled.

andypiper, 2024-12-17:
On 15.2 if I disable the default macOS firewall, the camera works - I can access it in the Preview in Bambu Studio (although Go Live is not working). If I enable the firewall, it shows this error about the camera malfunctioning.

garglebutt, 2024-12-20:
Same issue.

DrMaRV, 2024-12-21:
It's 100% macOS internal firewall, a have the same issue. As I've disabled macOS firewall in network settings the camera stream worked instantly.
My system:
iMac late 2017
macOS 15.2
BambuLab Studio 1.10.1.50

garglebutt, 2024-12-21:
So Bambu Studio needs to add code to seek appropriate permissions. Not hard to do given every other piece of software I have seems to manage.

bobg2023, 2024-12-21:
This ticket is closed I think a new one needs to be created. My firewall is permitting all inbound traffic to bambustudio. Perhaps the video stream is a different process?

andypiper, 2024-12-22:
I wondered the same thing (the stream being a different process). I added all of the Bambu processes/binaries to the firewall allowlist and it did not help, and I cannot tell from the logs what the issue is here.

bobg2023, 2024-12-23:
I just got some weird response from bambu support telling me to enable webstorm which is a programming IDE in local network access permissions.  I don’t even have that installed, what on earth does that have to do with the live view

martin2149, 2024-12-23:
I have a support ticket open for this issue as well, Apple M2 pro, Sequoia 15.2. 
It was working as expected and suddenly stopped, I can view camera on iPhone with no issue

dmrolyat, 2024-12-24:
I have same problem. Disabling firewall immediately resolves the camera problem

pranasziaukas, 2024-12-27:
Same here, the camera only works if the macOS Firewall is disabled, which I'm not willing to do.

bobg2023, 2024-12-27:
> Same here, the camera only works if the macOS Firewall is disabled, which I'm not willing to do.

If you haven't yet please raise a ticket with bambu labs support.

FlyMarc2, 2024-12-29:
> Same here, the camera only works if the macOS Firewall is disabled, which I'm not willing to do.

Same thing for the access to the MicroSD Card. (MacBook Pro 2018 (intel) and MAC OS Sequoia 15.2)

Made a support ticket. I hope they will solve the problem soon!

c0desurfer, 2024-12-30:
Why is this closed? This issue still persists.

bobg2023, 2024-12-31:
> Why is this closed? This issue still persists.

Because someone said turning off the firewall fiixes it and someone with little intelligence thought that was an appropriate fix

c0desurfer, 2025-01-01:
At least this one is open and it's the same issue.
https://github.com/bambulab/BambuStudio/issues/5642

Kpoole4u, 2025-01-01:
I lost video as well and now have it back....! 

I went into System Settings / Privacy & Security / Local Network. I turned off BambuStudio and then immediately back on. It is now working. I am using macOS Sequoia 15.1.1

bobg2023, 2025-01-01:
I've tried turning it off an on again more than once and it doesn't fix it

IOIO72, 2025-01-02:
@Mr-121 I think, this issue should be re-opened.

c0desurfer, 2025-01-02:
> I lost video as well and now have it back....!
> 
> I went into System Settings / Privacy & Security / Local Network. I turned off BambuStudio and then immediately back on. It is now working. I am using macOS Sequoia 15.1.1

Yes, but after updating to 15.2 (24C101) this doesn't work anymore.

kocour, 2025-01-03:
I can confirm that the same problem is on MBA M3 with 15.2 (24C101). Turning off integrated firewall will temporary fix the problem, but it's not solution at all, just absolutely terrible workaround. Obviously the problem is in the application itself and it should be updated for compatibility with 15.2

guinnberg, 2025-01-08:
Something odd is happening here. Yesterday it wasn't working as usual, I accidentally opened a second instance of Bambu Studio and kept working for the whole day. Now today I opened Bambu Studio again and no matter what I do, it doesn't work

just-mitchin, 2025-01-09:
Having the same problem after updating to 15.2 on a MBP M3. The only thing that works is to turn off the firewall. No issues on my iPhone app or Windows based PC.

bobg2023, 2025-01-09:
> Having the same problem after updating to 15.2 on a MBP M3. The only thing that works is to turn off the firewall. No issues on my iPhone app or Windows based PC.

Make sure you raise a support request with Bambu if you haven’t

FlyMarc2, 2025-01-12:
I have received another reply from Bambulab: 

> Hello,
> We apologize for the unpleasant experience. Could you please also disable any VPN or content filter you are using to see if there is any progress? There are reports that the Little Snitch filter may be causing this issue
> [image]
> Please feel free to contact us if you have any further questions.
> Best regards,
> Bambu Lab Customer Support
> 

It worked for me. Instead of Little Snitch, I have Trip Mode. After deactivating it, I had access to the live image and the SD card again. This is not a permanent solution either, but it is definitely better than switching off the firewall.
[image]

just-mitchin, 2025-01-12:
> I have received another reply from Bambulab:
> 
> > Hello,
> > We apologize for the unpleasant experience. Could you please also disable any VPN or content filter you are using to see if there is any progress? There are reports that the Little Snitch filter may be causing this issue
> > ![Bild 12 01 25 um 11 14](https://private-user-images.githubusercontent.com/128502701/402347835-e950c8c4-b1be-465b-b3bc-3ce2033af423.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY3MDk3NTYsIm5iZiI6MTczNjcwOTQ1NiwicGF0aCI6Ii8xMjg1MDI3MDEvNDAyMzQ3ODM1LWU5NTBjOGM0LWIxYmUtNDY1Yi1iM2JjLTNjZTIwMzNhZjQyMy5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDExMlQxOTE3MzZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yMWJhMjE1YzIyNjY2NGFjYjRlMjdmZGU4YjMxYmYyOGJlMDk2NGRiYWExYzc4ZDUwZGZmZTkwNmM1YjNhYmQ1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.cXVFFvhgxkVVwaQLFO6OODYjRA9TCSFwd_2lRqmI9Wo)
> > Please feel free to contact us if you have any further questions.
> > Best regards,
> > Bambu Lab Customer Support
> 
> It worked for me. Instead of Little Snitch, I have Trip Mode. After deactivating it, I had access to the live image and the SD card again. This is not a permanent solution either, but it is definitely better than switching off the firewall. <img alt="Bildschirmfoto 2025-01-12 um 10 35 04" width="813" src="https://private-user-images.githubusercontent.com/128502701/402348143-2d89863b-c288-4764-9eea-da0dd81a11c9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY3MDk3NTYsIm5iZiI6MTczNjcwOTQ1NiwicGF0aCI6Ii8xMjg1MDI3MDEvNDAyMzQ4MTQzLTJkODk4NjNiLWMyODgtNDc2NC05ZWVhLWRhMGRkODFhMTFjOS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDExMlQxOTE3MzZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05ZDIxYmIxZmRhZThmNDNlOTdmM2M4YzdlYmVlZGM1Y2EzNzBhMzI3NTQ1YzFlNjEzNzdhZGMyZjkwNGZmZDc2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.xBNSRaEBZHXLyQFNHGUblOitLWCM0tzDE2kEsSNUpkY">

Thanks! This worked for me. I have a Cisco VPN  installed and it now works with it being disabled.

IOIO72, 2025-01-13:
> I have received another reply from Bambulab:
>
>
> It worked for me. Instead of Little Snitch, I have Trip Mode. After deactivating it, I had access to the live image and the SD card again. This is not [...]

That was it! I had the TripMode filter active, even though the TripMode app was deleted.

garglebutt, 2025-02-08:
This is resolved with MacOS 15.3.