bambulab/BambuStudio #4920: Make function from Fusion 360 does not work anymore

jwortelboer, 2024-09-30:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 12.7.6

### Printer

Bambu Lab P1S with AMS

### How to reproduce

When using the Make function in Fusion 360, the design is not always (only the first time) transferred to Bambu Studio. 
Only when Bambu Studio is not running and Fusion 360 launches Bambu Studio through the Make function, every thing work as expected.

Be advised, this was working perfectly on the previous combination of and older Fusion 360 / Bambu Studio. Unfortunately i cannot reinstall an older version to find out which one is the issue. I am filing an issue here, hoping the good people from Bambu can investigate this, and maybe wordt case file this as a known issue from Fusion 360? 

### Actual results

When Bambu Studio is already running, the Make function does not transfer the file to Bambu Studio. I do see that Bambu Studio is activated in some way as it becomes active and gets the focus.

### Expected results

The normal result was that the design from Fusion 360 was send to Bambu Studio and would be presented on the plate as if it was added manually.

### Project file & Debug log uploads

This was verified on serveral systems. This is very likely introduced in the latest update from Fusion 360 or from Bambu Studio. Before (a week ago) this would work perfectly. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

owocowyy, 2024-09-30:
The same goes for "3D Print" function

jwortelboer, 2024-10-03:
I just installed the latest release from Fusion 360 that was released to day (Okt 3) and everything seems works again !
So it was a Fusion bug after all...
Closing this issue..

closed 2024-10-03 by jwortelboer