bambulab/BambuStudio #4921: Filament quantity display error

d-hesse, 2024-09-30:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1C

### How to reproduce

Slice all plates

### Actual results

Support filament shows the amount of model filament
[image]

### Expected results

Support filament shows the amount of support filament

### Project file & Debug log uploads

Sorry can't paste. Project is a secred prototype

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MackBambu, 2024-10-09:
Could you tell me what language you use ? Thanks

d-hesse, 2024-10-10:
> Could you tell me what language you use ? Thanks

German