bambulab/BambuStudio #4924: Set number of top/bottom layers in color

owocowyy, 2024-10-01:
I have a model with four top and four bottom solid layers. Both sides are painted using 3 colors. Right now I have to print all of those layers in colors. What I would like to do is to only print first two bottom layers in color and then following two layers in a single black color. The same goes for top layers. I only want to keep last two layers in color.

acabtp, 2024-10-06:
top is flat?  you can do this with color painting.  choose color painting, use the height range, set it to (your object height - 4 x the layer height), then mouse over the object and set the "bottom" to (2 x the layer height).  e.g. for a 10mm tall object and .2mm layers, set the height range to 9.2 and bottom to .4.

owocowyy, 2024-10-07:
Yes, my object is flat. I didn't get you explanation :/ Could you try this on this model? I want to keep the colors but instead of painting all 5 top layers in white, red and black I only want last two layers to keep the color (without decreasing the number of top layers). I don't care about colors inside the model. I want to decrease the amount of filament changes.

[set_number_of_top_bottom_layers_in_color.3mf.zip](https://github.com/user-attachments/files/17276844/set_number_of_top_bottom_layers_in_color.3mf.zip)