bambulab/BambuStudio #4926: Top Z Support Distance Ignored While Using Adaptive Layer Height with Tree Supports

creativedamages, 2024-10-02:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.0 & macOS 14.X

### Printer

P1P with upgrade kit, P1S

### How to reproduce

Select object and run adaptive layer height. Enable support. Select Tree (Auto). (Any Style). Slice. 

### Actual results

The Z height distance is ignored.

### Expected results

Z Height should be seen when sliced and printed. 

### Project file & Debug log uploads

Any file gives same result. See images:

Slice with tree supports (adaptive layer height)
[image]
Slice with normal supports (adaptive layer height)
[image]
Slice with tree supports (standard layer height)
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tangketan, 2024-10-08:
Please upload the 3mf for diagnosis.

Marclin54, 2024-10-08:
Same here...had to roll back to v1.8 for now...and this is all over the forums as well.

creativedamages, 2024-10-08:
This is file agnostic. Any file loaded has this issue.

tangketan, 2024-10-10:
I found a 3mf which may reproduce the issue you mentioned. Indeed there is some issue with adaptive layer height + tree support + very large top z distance. That's a case I didn't test enough. I'll fix it in the next version. For now, you can try to use a relatively smaller top z distance value. From my experience, 0.5mm is already a very large value that barely works. 5mm top z distance will make support meaningless.

Marclin54, 2024-11-12:
Don't see anything aboout this issue in the release notes for 1.10.0...has it been fixed?

Johnjr00, 2025-11-02:
@tangketan 

Its over a year later now and this issue is still in the latest version of Bambu Studio... Will this ever be fixed?