bambulab/BambuStudio #4927: ASA-CF not recognised by AMS properly

MechInnovation, 2024-10-02:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab X1C

### How to reproduce

Try and load Bambu Lab ASA-CF in the AMS. 

### Actual results

The AMS goes through it's usual routine of loading the filament and then briefly displays (about 7 seconds) ASA-CF correctly, but then the user interface in the printer goes back to a blank slot as if it doesn't know what is loaded.

https://github.com/user-attachments/assets/a189b020-9c91-4efd-b51e-e4558c8ee8e5

### Expected results

It should show the material as normal. 

Also there is no profile in the materials within Studio for the ASA-CF. There seems to be very little information online about ASA-CF and now it makes me very nervous about using it.

I've created a Project file and log file because I can't submit a support ticket without it but they are relevant at this stage because I can't get past the material loading stage. I have also tried loading the filament directly being fed from a spool drier (not via the AMS) and although I can select ASA-CF it again only shows it for a few seconds and then says it doesn't know what it is.

What's going on?

### Project file & Debug log uploads

[debug_Wed_Oct_02_10_11_02_56240.log.zip](https://github.com/user-attachments/files/17227544/debug_Wed_Oct_02_10_11_02_56240.log.zip)
[Bambu Cube.zip](https://github.com/user-attachments/files/17227550/Bambu.Cube.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

MechInnovation, 2024-10-02:
Please can you sort this out? It makes it look like you want to supply the material but then it's not supported by the ecosystem. I'm very happy to purchase only Bambu filaments, but they need support.

SiliciumMan, 2024-10-04:
Please navigate to the "Update Configuration" section to ensure your settings are up-to-date. Whenever new materials are released, we will push updates, so if you don't update, you'll be prompted the next time you open the app.

[image]

Making sure your configuration is current will allow you to access the latest features and capabilities. Please take a moment to review and update your settings as needed.
============================
I asked the same question with Bambu customer service. This is their answer.
I checked update, the software is newest. And the ASA-CF profile is not availble in github just now. But it really has a configuration  update in the right below corner.
I click it and get the update. Now it works!

MechInnovation, 2024-10-04:
Thanks for your reply. I have the latest version of Studio (on PC). Both my X1C and AMS appear to be on the latest version, but I don't get the 'Configuration can update now' box popping up like you have shown in your image. How do I get that? Any help would be very much welcomed :-). I've only had the print for 2 weeks and love it, but this has got me a bit stuck.

MechInnovation, 2024-10-07:
> Please navigate to the "Update Configuration" section to ensure your settings are up-to-date. Whenever new materials are released, we will push updates, so if you don't update, you'll be prompted [...]

Mine is working now but I had to reinstall BS and the printer firmware.

XunZhangBambu, 2024-10-15:
@MechInnovation  Hello,has your issue been resolved ? To ensure filament recognition, you need to make sure that both the studio and the printer have received the new materials package.

MechInnovation, 2024-10-15:
Yes this has been solved now. I u installed everything and the reinstalled.

Kind regards

Simon

Simon Stone

MechInnovation Limited

7 Enright Close

Leamington Spa

Warwickshire

CV32 6SQ

Mob: +44 (0) 7919 533515

***@***.******@***.***>

www.mechscan.co.uk<http://www.mechscan.co.uk/>

closed 2024-10-17 by ttt888zzz