bambulab/BambuStudio #4933: Wrongly summarized support filament usage when slicing multiple pla

emil-nasso, 2024-10-03:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Mac

### Printer

A1

### How to reproduce

Sliced all plates of this https://makerworld.com/en/models/676112#profileId-604218

i removed some models from the plates (duplicates/color variants) and sliced all plates to get an idea of total filament usage.

The support filament usage seem to be summarizing the model filament usage values.

### Actual results

970g

### Expected results

A lot less

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

emil-nasso, 2024-10-03:
[PUMP2000V1.0.3mf.zip](https://github.com/user-attachments/files/17241431/PUMP2000V1.0.3mf.zip)

DrOetcker, 2024-10-03:
I can confirm this error as I just wanted to create an Issue for it.

After manual rearrange & Filament change of the official A1 Top Mount Model in Studio, the estimation of the "Slice all" option is wrong. The total calcuation for the Model itself is correct, but the total calculation for the support is wrong and identical to the Model.

Bambu Studio Version: 
1.9.7.52

OS: 
Windows 10 Pro

Printer: 
A1

[AMS+Lite+Top+Mount-Final - Wrong Filament calc on slice all.zip](https://github.com/user-attachments/files/17242767/AMS%2BLite%2BTop%2BMount-Final.-.Wrong.Filament.calc.on.slice.all.zip)

Greetings
Oetcker

[image]

LiZ-Li-BBL, 2024-10-09:
This issue has been fixed internally, please wait for our update.

closed 2024-10-10 by emil-nasso

ttt888zzz, 2024-10-21:
Please update the newest version，https://github.com/bambulab/BambuStudio/releases
[image]