bambulab/BambuStudio #4934: Support issue with adaptive layer height

molnard, 2024-10-03:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

win11

### Printer

X1C

### How to reproduce

Slice the model.

### Actual results

One support column is missing. 

[image]

### Expected results

Everything is supported. 

### Project file & Debug log uploads

[tara_adaptive.zip](https://github.com/user-attachments/files/17244546/tara_adaptive.zip)
[debug_Thu_Oct_03_14_43_08_11236.log.zip](https://github.com/user-attachments/files/17244579/debug_Thu_Oct_03_14_43_08_11236.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

tangketan, 2024-10-08:
The support column is not missing. Instead, it touches the model surface and stops there. It should be printed without problem, except a little artifacts on the touched surface.

molnard, 2024-10-09:
> The support column is not missing. Instead, it touches the model surface and stops there. It should be printed without problem, except a little artifacts on the touched surface.

Interesting, in what version of Bambu lab did you load the model that I have provided above? 

For me is is clearly hanging in the air and the print will be a scrap. 

[image]

molnard, 2024-10-09:
I have also notice a many other issues with the support when you apply adaptive layer height. It is not generated at all or just generated partially like the example above. I can provide more models where the issue is visible if required.

ttt888zzz, 2024-10-12:
HI [molnard](https://github.com/molnard):The actual result looks fine. 
[image]
[image]

molnard, 2024-10-29:
Thanks for the reply. Look at layer 1409. The support created in the air. 

[image]

Practically the issue might resolve itself because other parts of the model is close so the support can build up - but only in this case and only acceptable in low quality prints. What I see here is that the algorithm is trying to build the tree down there to the build plate, according to my setting but at some point the tree generation alg fails when adaptive layer is used and only a small chunk created. I encountered the same issue at multiple models and that made the support unusable.