bambulab/BambuStudio #4937: Pressure Advance does not show in Filament settings

NVNDO, 2024-10-04:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10

### Printer

P1S

### How to reproduce

Go into any Filament setting. "Pressure Advance" is no longer shown. 
Search for "Pressure Advance" using the Search icon on the top right corner.
It now shows the 2 options, but regardless of how many times you click, they do not add or show in settigns. 
From how they're listed in the search-window, they should be visible under "Basic information" but they don't.  

[image]
[image]

### Actual results

Pressure Advance setting in no longer available.. visible...

### Expected results

Be able to activate/deactivate, adjust Pressure Advance in the Filament settings - Basic Information tap. 

### Project file & Debug log uploads

No files necessary

### Checklist of files to include

- [ ] Log file
- [X] Project file

NVNDO, 2024-10-05:
> Pressure advance has **never** been included in the Studio filament settings. It is accessible in Studio only from the Device tab.
> 
> If you want proper control over pressure advance via the filament preset, use OrcaSlicer.

That's exactly the irony I was going for. 

You can see the PA-values in  "Compare Presets" 
"Search" gives you a preview for both settings *shown above* (that should be easily accessible in settings) though they're hidden to the user. I know they're not there - but they should. That's also the reason I've been using Orca-Slicer to this point. (which start to get annoying when you regularly upload designs to Makerworld - as you have to constantly change between them)

You guys @Bambulab have done so many changes (for good) recently, that implementing this would be a game changer for many as we already know our PA-values and don't want to redo all the tests for each filament (as necessary for right now) to get them set.

Thank you so much in advance ✌

RyanEwen, 2024-10-09:
@NVNDO this has been requested before by myself and others but Bambu pushed back, as they fear that the PA values may differ per filament **per printer**. Their solution is to have you go to the Calibration Tab then click Manage Result and save your PA results there.

It's better than nothing but not ideal as you can only create profiles there for the default filament types built-into the software (so if you have added a BrandABC profile yourself, it's not going to be in the list, and you have to choose Generic instead). Fortunately, you can name the profiles whatever you want, just be aware that you'll only see profiles for the selected brand when editing an AMS slot on the Device view.

NVNDO, 2024-10-12:
Hey @RyanEwen, thank you so much for reaching out and giving me their answer - which they could have given here as well. 

I had not thought about the fact that PA would be the same for all printers and might differ. Therefore their approach is the way to go. Pretty damn annoying having to recalibrate everything again and again, but I do get the point now. 

Thank you so much for your time.

RyanEwen, 2024-10-12:
I'm skeptical about the PA values differing between machines (I have 5 and they're always the same) so I would much prefer if the filament profiles are least held a default PA value that could be overridden. Similar to how 0.020 is assumed if you don't do a calibration or if you don't choose one on the Device view. 

Also annoying that we cannot just type a number on the Device view anymore. We are forced to create a profile and choose a profile. 

Further, there's currently a bug where these PA profiles don't show up on the Device view after changing devices.

NVNDO, 2024-10-12:
We all assumed Bambu is a copy of Orca, but why did they leave all the good features behind?! 

Some things I'll never understand...

RyanEwen, 2024-10-12:
It's the other way around - Orca is a fork of Bambu