bambulab/BambuStudio #4938: BambuStudio does not work on my notebook

adras, 2024-10-04:
### Bambu Studio Version

latest

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

DELL XPS 15

### Printer

Bambulab X1 Carbon

### How to reproduce

1. Open Bambustudio, go to the device tab, connect the camera in lan mode
2. Put notebook into standby, wait 10 minutes
3. open notebook again, bambustudio will be disconnected
4. close bambustudio

1. Open Bambustudio, print something

### Actual results

In the first case the password will be deleted from the configuration and will be needed to be reentered
In the second case there will be an ftp upload error at 30 or 70 percent with a reason of: timeout

### Expected results

1. Bambustudio not forgetting the password because of disconnection
2. Successful FTP uploads on an Dell XPS 15 

### Project file & Debug log uploads

no files required

### Checklist of files to include

- [X] Log file
- [X] Project file

mintdaniel42, 2024-10-05:
1. It would be great if you'd provide a specific version rather than "latest" 
2. The problem might be caused by the faulty operations system you are using. Windows itself is a source of problems. This can be solved by installing a proper operation system like Linux

walterwongbbl, 2024-10-11:
@adras 
We are currently investigating this issue