bambulab/BambuStudio #4939: nozzle always goes to wipe for no reason

Makytash, 2024-10-05:
### Bambu Studio Version

latest as 05.10.24

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10

### Additional system information

[image]
[image]

### Printer

a1 and a1 mini

### How to reproduce

[image]
[image]

the travel makes the print looks really bad and takes 11min more for no reason, I've tried on x1 carbon same slice the problem wasn't there.

### Actual results

the nozzle always goes to wipe for no reason, Ive dissabled prime tower to but still, I have no Idea how to fix this. please help!!

### Expected results

fix issue, or please tell me what can I do to make it work

### Project file & Debug log uploads

it has the same problem on everything as long as its on a1 or a1 mini 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Leopardo1967, 2024-10-06:
Hello, I have had exactly the same problem since yesterday, so I can't print anything with any guarantee of success.
If I load a previous job, it works fine, but every new project comes out as shown in the images in this thread.
[image]

Leopardo1967, 2024-10-06:
this is my PC status
[image]

Leopardo1967, 2024-10-06:
[image]

Leopardo1967, 2024-10-06:
Hi,
I just try to work with my wife's Pc (HP with 2 years old and win 11) and find the same isuue, also when I try to use othe bambu lab printer there is no any problem anly that I can't send the STL by other diferent than A1 filter, so by the moment I can't print nothing.
Please don't left me along with tihs problem.
Thanks

Leopardo1967, 2024-10-07:
Hi,
I have found that if you choose print by object then it doesn't happen.

ttt888zzz, 2024-10-08:
HI [Leopardo1967](https://github.com/Leopardo1967)：Please update the newest version，https://bambulab.cn/zh-cn/download/studio

Leopardo1967, 2024-10-08:
Hi, thanks for your support, the last version also fail, by the moment I have no solution.

XunZhangBambu, 2024-10-09:
@Leopardo1967 
Hello, what you see is the nozzle moving away from the hotbed to take timelapse photos. During actual printing, you can uncheck timelapse, and the nozzle will not move to take photos at that position. The timelapse behavior varies for different machine models. For specific details, you can refer to our wiki at https://wiki.bambulab.com/en/software/bambu-studio/Timelapse.

Leopardo1967, 2024-10-09:
[image]

Leopardo1967, 2024-10-09:
https://github.com/user-attachments/assets/9c0eaf49-b8df-403f-aaf3-7c0fc60bf24c

Leopardo1967, 2024-10-09:
Hello,
I understand what you are saying, believe me, but you cannot deactivate the time lapse movements in my settings when you do not use them and they are supposed to be deactivated, I have tried everything, what you tell me too, I am only printing because I found where to delete the code in the file that contains the G-CODE for time lapse.
Sincerely

Leopardo1967, 2024-10-10:
Today all is fine, I test it and this isuue has disappear so I'm happy again.
Thks a lot

XunZhangBambu, 2024-10-11:
@Leopardo1967 I am very happy that your issue has been resolved.