bambulab/BambuStudio #4940: Skip button stops the support material change from happening

BTipold, 2024-10-05:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

Winsows 11 Version 10.0.22631 Build 22631

### Additional system information

https://imgur.com/xxz7Nl2

OS Name	Microsoft Windows 11 Home
Version	10.0.22631 Build 22631
Other OS Description 	Not Available
OS Manufacturer	Microsoft Corporation
System Name	DESKTOP
System Manufacturer	ASUS
System Model	System Product Name
System Type	x64-based PC
System SKU	SKU
Processor	AMD Ryzen 7 5800X 8-Core Processor, 3801 Mhz, 8 Core(s), 16 Logical Processor(s)
BIOS Version/Date	American Megatrends Inc. 3211, 2021-08-10
SMBIOS Version	3.3
Embedded Controller Version	255.255
BIOS Mode	UEFI
BaseBoard Manufacturer	ASUSTeK COMPUTER INC.
BaseBoard Product	TUF GAMING B450M-PRO S
BaseBoard Version	Rev X.0x
Platform Role	Desktop
Secure Boot State	On
PCR7 Configuration	Elevation Required to View
Windows Directory	C:\WINDOWS
System Directory	C:\WINDOWS\system32
Boot Device	\Device\HarddiskVolume2
Locale	United States
Hardware Abstraction Layer	Version = "10.0.22621.2506"
User Name	DESKTOP\Brian
Time Zone	Eastern Daylight Time
Installed Physical Memory (RAM)	64.0 GB
Total Physical Memory	63.9 GB
Available Physical Memory	48.3 GB
Total Virtual Memory	92.9 GB
Available Virtual Memory	63.9 GB
Page File Space	29.0 GB
Page File	C:\pagefile.sys
Kernel DMA Protection	Off
Virtualization-based security	Not enabled
Windows Defender Application Control policy	Enforced
Windows Defender Application Control user mode policy	Off
Device Encryption Support	Elevation Required to View
Hyper-V - VM Monitor Mode Extensions	Yes
Hyper-V - Second Level Address Translation Extensions	Yes
Hyper-V - Virtualization Enabled in Firmware	No
Hyper-V - Data Execution Protection	Yes

### Printer

P1S, AMS

### How to reproduce

[Preconditions]
Default unless specified below:
- support: enabled
- support base: (3) PLA
- support raft interface: (4) PVA
- top z distance: 0
- interface pattern: rectilinear
- top interface spacing: 0
- independent support layer height: No
- walls: 6
- infill: 40
- infill pattern: gyroid
- prime tower: No

[steps]
1. slice and print
2. allow some of the print to go, allow a few filament changes
3. using the companion app version 2.14.0 on iphone 14, press SKIP and select 1/4 parts

### Actual results

- remaining 3 models complete successfully
- AMS does not switch anymore, remainder of the print is done in PLA only, including the support interface

https://github.com/user-attachments/assets/dd85fe0a-5bfd-4342-b1e5-f751a1ffc504

https://streamable.com/gbq7n1

### Expected results

- remaining 3 models complete successfully
- AMS switches to PVA for support interface as expected 

### Project file & Debug log uploads

[for-reproduction.zip](https://github.com/user-attachments/files/17264711/for-reproduction.zip)
[debug_Thu_Sep_26_21_41_33_196.log.zip](https://github.com/user-attachments/files/17264717/debug_Thu_Sep_26_21_41_33_196.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

ttt888zzz, 2024-10-08:
HI [BTipold](https://github.com/BTipold)
1.The yellow filement did't work，is it right？The expected result is 2 colors，the Actual result is 1 color，right？
[image]
2.about “iphone 14, press SKIP ”：Could you Please provide a gif or photo，we will check it

BTipold, 2024-10-08:
Hi @ttt888zzz 

1. Correct, the yellow filament worked at first. But then after I pressed skip, it seemed to have stopped using the yellow. It just printed everything in PLA past that point.

2. Unfortunately I had guests staying in my condo so I did a bit of a deep clean and threw out the parts. But I do have a photo I took.
[image]
See the highlighted area that clearly doesn't have the support material. It should be there.
By memory, I was recall being surprised that I hadn't heard the filament change happen in a while. I didn't think much of it. Then when I tried to remove the supports, I found that it was all PLA for the second half up. You can also clearly see this in the streamable. https://streamable.com/gbq7n1 the clear PVA filament (yellow in slicer) goes down for the first half. But after the the skip button is pressed (roughly 0.750s), the clear filament stops being used. Even though its in the slicer. Its should be used for support interface of those two holes near the centre of the frame.

BTipold, 2024-10-08:
https://streamable.com/oq3r9t

Here is a video of the skip button.
I can't press it now because I have an actual print running. I don't want to skip this one, its turning out well.

Can you guys try reproducing this on your end? If not I can try to reproduce tomorrow.

MartinChan3, 2024-10-30:
[1.10 beta2](https://github.com/bambulab/BambuStudio/releases/tag/v01.10.00.81) has solved this problem.

closed 2024-10-30 by zhimin-zeng-bambulab

BTipold, 2024-10-31:
Thank you for the prompt action on this, I appreciate your hard work!