bambulab/BambuStudio #4941: Shut down printer in case of fault

StuTheDog, 2024-10-05:
Occasionally the printer will stop due to a fault- spaghetti detected, first layer defects, filament problems etc.- awaiting action from the operator. Certain faults result in printer shutdown (bed and extruder cooling, fans shut down etc.), however some faults leave everything running (1st layer defects for example). This behavior, although inconsistent, is acceptable WHEN THE OPERATOR IS AVAILABLE.

However if I am not available to fix the problem quickly (for example I'm sleeping during an overnight print job) leaving the printer running is wasteful and potentially shortens the lifetime of the printer.

I would suggest several changes- In your code create a default/global exception handler for all faults and always throw to this handler in the event of a fault.  Modify the exception handler to keep the printer running (fans and heaters left on (possibly exception of the extruder) for a short time (e.g. 5 minutes) after which shut everything down if there has been no response from the operator.  Add a preference to set the duration e.g. 0, 1, 5, 10 minutes for the exception to wait for user interaction.  And finally, include the internal light and camera in this group to be shut off.

acabtp, 2024-10-06:
please do not implement this feature.  most users would prefer to fix the issue and resume the print job when they wake up, return from work, etc.  the user who suggested this can simply run their prints WHEN THE OPERATOR IS AVAILABLE and not when they aren't there, and the rest of us can have the printer keep working as it does now.

StuTheDog, 2024-10-10:
I believe the commenter misunderstands my request. I am not suggesting to cancel the print job. I am suggesting to store all settings as is done now, send notifications to the user as is done now, wait a period of time (5 minutes or other set by preferences), if no response by then shut down bed and nozzle heaters, shut down cooling fans and the interior light if active.  Once the user does respond, restore all settings, re-heat the heaters, start the fans, turn on the light if applicable and resume printing.  My request is simply to stop running all devices while awaiting a long period of time for a user who might be away.

acabtp, 2024-10-11:
Shutting down the bed heater ruins the print.  Please do not implement this feature.  The original poster can just run their prints while they are available to attend to the machine.