bambulab/BambuStudio #4943: [Linux] Dark mode objects and Selected tab unreadable colours

AkashCherukuri, 2024-10-05:
### Bambu Studio Version

1.97.52

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 24.04 - Noble Numbat

### Printer

Bambulab A1

### How to reproduce

Turn on dark mode in slicer.

### Actual results

[image]

There are two issues here:
1. The text color of the objects is unreadable against the background
2. Text color of the selected tab (`Frequent` in attached screenshot) is darkened, possibly from Light mode. This causes the selected tab's name to be unreadable against the dark background

### Expected results

The text should be light-colored

### Project file & Debug log uploads

Not required.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

AkashCherukuri, 2024-10-05:
I could try fixing this myself and making a PR, since its pretty simple to do.

eduardogsilva, 2024-10-06:
Hello, I can confirm that I also have this issue.
A lot of unreadable or very hard to read texts.

ubuntu 24.04 and bambu studio 1.9.7.52.

Thank you! 😊

MackBambu, 2024-10-11:
Does dark mode in Ubuntu follow system settings properly?

eduardogsilva, 2024-10-11:
> Does dark mode in Ubuntu follow system settings properly?

I believe so. It's Ubuntu. It also uses Gnome desktop environment. 
So far, bambulab is the only application that I use with color issues.

Those issues also happens on the light mode.
Light mode have less issues, but they are still there.

Screenshot below 

[image]

eduardogsilva, 2024-10-11:
Dark mode have a lot of more issues, just check in the object/filament column in the left

[image]

al7810, 2025-01-28:
Same issue: application is closing dialog text ist unreadable ("white on white"). Linux Mint 21.3 Cinnamon

MackBambu, 2025-03-26:
Fixed in 2.0

closed 2025-03-26 by MackBambu