bambulab/BambuStudio #4944: Allow the modification of the default System Filament Profiles or a way to map/redirect them to custom User Filament Profiles.

sheehand, 2024-10-05:
**Is your feature request related to a problem? Please describe.**
Every time I press the filament Resync button in Bambu Studio, or I load a new filament in the AMS using any of the System preset filament profiles (such as manually selecting Generic Silk PLA or using the auto-detected Bambu Lab profiles) and then I press the Sync button in Bambu Studio, the filament profiles in Bambu Studio default back to the System Filament Profiles.
For the record I have had to Resync filaments from time to time in Bambu Studio when the Sync button isn't picking up just the filament changes in the current project for whatever reason, and I have to select Resync to fix that issue which resets all custom filament profile selections back to the default ones.

The problem with the default System Filament Profiles is that at a minimum I need to modify their bed temperatures to make sure the model(s) I am printing don't have bed adhesions issues. I.E. The default bed temps don't work in my environment and models often come loose using them. To address this, I have created modified User Preset Filament Profiles for every Bambu Lab filament profile, as well as just about every Generic filament profile, and saved those modified versions with "- Tweaked" on the end of the existing name. For example, I modified "Generic PLA" by upping the bed temperatures and saved it a User Filament Profile named "Generic PLA - Tweaked". 

As long as I remember to manually override each and every individually chosen default filament profile selection in Bambu Studio with my custom profile, I don't have bed adhesion issues due to the wrong temperature. The problem is every so often I miss swapping "Generic PLA Silk" --> "Generic PLA Silk - Tweaked" or "Bambu PETG Translucent" --> "Bambu PETG Translucent - Tweaked", and then the model will have printing issues as a result. I'm not perfect and sometimes I miss 1 or more System --> User filament profile changes, especially when it's 3AM and I am trying to get a print to fire off before I go to bed WAY past my bedtime. 

**Describe the solution you'd like**
I would like to be able to modify the default System Filament Profiles to use the required tweaks (mostly increased bed temp tweaks), but I realize this is not necessarily something people would be willing to go for as system defaults are often considered sacrosanct (and I understand why). FYI, I looked on my computer's the file system and I can only find the User Filament Profiles under the Bambu Studio folder, so I don't think it's possible to modify the defaults even if I wanted to.

Therefore, as alternative I would like the ability to tell Bambu Studio "every time you think you should use the 'Bambu PETG Translucent' filament profile, select the 'Bambu PETG Translucent - Tweaked' user filament profile instead" through some sort of automatic mapping or redirect in Bambu Studio. I.E. A mapping file that says:
"Bambu PETG Translucent" --> "Bambu PETG Translucent - Tweaked"
"Generic PLA Silk" --> "Generic PLA Silk - Tweaked"
"Bambu PLA Basic" --> "Bambu PLA Basic - Tweaked"
etc.

Either approach would ensure that at 3AM when I swap one filament for another on the printer and go to Bambu Studio and refresh the filaments, the correct tweaked filament profile will be used for the type of filament automatically detected (Bambu) or manually set (Generic) in my AMS. This would also be one less thing I have to check and recheck for each and every new 3D print model I download and try to print, as I mentioned above the Sync button often does not pull in the correct filaments from my printer which causes me to use Resync which reverts User Filament Profiles back to the System Filament Profiles.

**Describe alternatives you've considered**
I tried to create Custom Filament Profiles, but we are blocked from using the vendor names of Bambu or Generic. I understand I could create my own names like "BL" and "Generic2" to try and work around these name blocks so I can add my own filament names to the filament selection on my printer, but that doesn't help for two reasons:
1. The Bambu Lab filament autodetection in the AMS will still default back to the System Filament Profile in Bambu Studio as I outlined above, and each time that happens I go the extra step and remember to manually override the filament selection to the custom one I created, which totally defeats the purpose of the Bambu Lab RFID automatic filament detection IMHO. I.E. It's either fully automated or it's 90% automated and if you forget to do the last 10% your print could fail due to adhesion issues.
2. I don't want to create BL and Generic2 filament profiles because I feel the filament list on my X1C will get too confusing (Generic vs Generic2) over time not to mention start to make the manual filament selection lists in the AMS crowded if I do ever create a true vendor specific filament profile for some reason. This is why I try to use the Generic profiles for just about every non-Bambu Lab filament because the filament pulldown list in Bambu Studio is already long and crowded enough as it is w/o adding more and more variations.

**Additional context**
Hopefully I have articulated the issue, the concern, and the ask clearly enough, but I do want to say that leveraging the automatic filament detection mechanism for Bambu Lab filament w/o having to manually override the filament selection every time at the printer and/or Bambu Studio, and using the existing Generic filament profiles for all non-Bambu Lab filament, is what has allowed my 11 and 8 year olds to be able to swap out filament w/o a lot of instructions or confusion. I would rather be able to override the settings in Bambu Studio or create a mapping file in Bambu Studio that I maintain, so my kids don't have to use more complex instructions to swap out filament as I think that would turn them off as being too complicated. Plus, again I'm human and make mistakes and would like to eliminate the "oh darn....I forgot to swap "Bambu PLA Basic" with "Bambu PLA Basic - Tweaked" on that last model I downloaded and printed from Makerworld... 

If you don't agree with my reasoning or logic, please don't flame me but instead offer constructive alternatives. I'm open to things I haven't considered, but I really think some sort of optional mapping file would solve the issue that I and several other makes I have talked to agree is about the only short coming to downloading and printing models using Bambu Studio.

TheRealHotFix, 2024-10-19:
Another failed ~11 hour print because I started it at like midnight and forgot to switch "Bambu PLA Basic" to "Bambu PLA Basic - Tweaked" amongst all the other tweaks and modifications (including hand washing the build plate with dish detergent) before hand.

There really needs to be a way to automate the change in the slicer and/or printer so the default bed temps for the pre-existing filament profiles can be bumped up from 55c to 65c.

I understand I can and should do this manually every time, but I am trying to simplify and automate the printing process just a little bit more w/o having to create duplicative filament profiles that I then have to manually select each time at the printer. I.E. Not have to manually override the Bambu Lab RFID selected filament profile at the printer or in the software, rather have the switch in bed temp be automated somehow.

papajin68, 2025-01-06:
I'm finding this to be a huge annoyance as well.  I just had a large flat print start curling up at the edges about 30 minutes in because I forgot to switch to my custom profile after swapping to a different color.  Another solution would be for Bambu to update the bed temp on the default profiles to something more reasonable... not sure why they have it set to 35 as that's going to cause a lot of PLA failures.  Even if they fixed that though, I think your feature request still has merit.

Personally I think they should allow us to directly override the default profiles however we please, but add a button to allow the user to revert to the system supplied default values at any time.  It can even highlight what has been modified like it does in the print settings with a button to revert just a single setting or all modified settings.  In most cases I use the same settings for each brand/type of filament so the need/desire for a custom profile is fairly rare for the things that I do -- In most cases, I'd much rather it just pick a profile I've defined when I put a Bambu filament in the AMS without my having to switch anything unless I'm doing something out of the ordinary that requires special settings.

chrisjohnson, 2025-01-24:
I would also love to see something along these lines. I find the default filament presets for my bambu filament to not be adequate either. Since the filament comes with an RFID tag that identifies the filament, and whenever I use a given type of filament, I'm going to want to use my chosen user preset for that filament type.

It seems like currently the system is RFID -> Select System Preset and it would be great if a mapping layer were added in between, where I can optionally specify a default alternate preset for a given RFID filament type. If I don't specify it will continue to fall through to the system preset (current behavior), but by setting the default alternative preset I can be sure that every roll of Bambu Basic PLA will map to the same user preset that I have decided works best for my prints

ArnoldStorm, 2025-03-19:
I strongly back the above suggestions.  I have the same issue with Bed Temp, but also the $.   All the default prices are in $US, but I'm in New Zealand, and when converting to $NZ it is almost double.   I can create custom profiles with the increased price, but always have to remember to swap out the default profile that the AMS selects.   One of the key features of the Bamnbu I liked was the RFID tag, to automatically select the correct profile.  But thats a bit usless if you cant edit the profiles.

Yegorich555, 2025-04-28:
Completely agree. It's nonsense that I can create custom profile for filament and for printer, but I can't setup only once by default or override existing presets. I need to not forget checking printer preset and every filament preset every time. It's annoying and not reliable at all.

geoffferg, 2025-05-30:
I strongly agree - very similar issues to those expressed above.

DanyloOliinykSSG, 2025-11-10:
upvote

Toggles18, 2025-11-16:
@DanyloOliinykSSG that works for creating a custom profile.  What @TheRealHotFix is saying that every time you sync your filament up from the AMS to the project, it defaults the filament profile to the "System Preset" for the brand you are using instead of the custom profile you have made.  So you have to manually click it and change it every time to the custom profile you want to use; it is just a pain to remember to do every project.  

What @TheRealHotFix is saying is they want a way to automatically have it go to the custom profile.

DanyloOliinykSSG, 2025-11-17:
@Toggles18 Not sure if i get it correctly but i have set custom profile filament to my AMS and it stays there persistently even after bambu studio/printer restart 🙄 When i create new project it defaults to my selected custom profile automatically.
The only thing happens to me is PA profile resets to default from time to time so i have to check it before (almost) every print.

[image]

[image]