bambulab/BambuStudio #4945: Multicolor printing no longer works when slicing in Bambu Studio 1.9.7.52 and printing on a non-updated X1C

buelojsh, 2024-10-05:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab X1 carbon

### How to reproduce

1. Paint model with multiple colors in Bambu Studio 1.9.7.52
2. Slice multicolor print
3. Export plate sliced file to MicroSD card.
4. Plug MicroSD card into X1C on firmware level 01.04.01.00 (unable to update because company's IT does not allow this printer to connect to internet)
5. Start print with AMS, Bed leveling, and flow calibration, but not timelapse

### Actual results

Print starts and is able to print. Pauses at each intended color change, goes to purge chute but does not purge or change filament. Makes purge tower and continues printing. Color never changes.

### Expected results

Print should change colors at each intended color change, instead of continuing with same color. Can verify it is not a problem with the AMS, as files sliced with older release print in multi-color with no issue

### Project file & Debug log uploads

[Uploading 2_part_turbo v2.gcode.3mf.zip…]()

### Checklist of files to include

- [ ] Log file
- [X] Project file

zhimin-zeng-bambulab, 2024-10-08:
@buelojsh Could you share your 3mf file? This 3mf hyperlink is a link to the current URL, not a document.

zhimin-zeng-bambulab, 2024-10-21:
@buelojsh  hello, could you share your 3mf file?

wizwho, 2024-10-22:
Thanks for getting back to me. I managed to tear out a connector socket on the extruder interface board trying to replace the hot end and am waiting for a replacement part to arrive before i can power up again to get the log file.

The problem is the same for any 3mf file, also for a new download of Bambu Studio.

Ted King