bambulab/BambuStudio #4948: New update no slicing

AdamheesePRINTS, 2024-10-06:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Ventura 13.6.9

### Additional system information

MacBook Pro 13" 2017 2,3 GHz Dual-Core Intel Core i5 8GB RAM

### Printer

Bambu Lab P1S

### How to reproduce

Updated the software and since then the slicing stops at 15%. Deleted the app and downloaded again and installed. Facing the same issue. Deleted software and downloaded the previous version and slicing works.

### Actual results

Slicing stops at 15% and never finishes

### Expected results

Continue slicing until printing is possible

### Project file & Debug log uploads

no log available

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ttt888zzz, 2024-10-08:
HI [AdamheesePRINTS](https://github.com/AdamheesePRINTS)：Could you please provide the source file，we will check it

AdamheesePRINTS, 2024-10-08:
Sorry for my late response.
This website/chat doesn't let me upload the file type dmg. 
It was downloaded through the app on my Mac. After starting the appp it told me there's an update available and I downloaded it from that link. I deleted the app after and downloaded it from the official Bambu lab website with the same issue.
This is the link:
https://github.com/bambulab/BambuStudio/releases/download/v01.09.07.52/Bambu_Studio_mac-v01.09.07.52-20240920143753.dmg

ttt888zzz, 2024-10-09:
Yes，the link is OK. https://github.com/bambulab/BambuStudio/releases
[image]

AdamheesePRINTS, 2024-10-09:
Ah of course. It is a project from maker lab and I just deleted th
[Selbstbew topf.3mf.zip](https://github.com/user-attachments/files/17309931/Selbstbew.topf.3mf.zip)
e prime tower. I hope it doesn't make any difference that I saved the file with previous Bambu lab version.

Thee website doesn't let me upload the file type .3mf so I uploaded a .zip file now.

zhimin-zeng-bambulab, 2024-10-21:
@AdamheesePRINTS  I tried slicing successfully, but it took a while. Could you have a try in 1.10 beta 1?

AdamheesePRINTS, 2024-10-21:
I installed the beta version and I was able to slice a small piece that I just printed recently and it worked. So I went to maker lab again to open the project: https://makerworld.com/de/models/500980#profileId-416030
This time I didn't change anything and just hit slice. The fan of the laptop starts to go crazy and slicing stops at 15%.
I opened another model that is big but not too crazy. This time it took a while and it went up to 50% or so. It needed maybe a minute to complete. Seems like the issue starts with big and complicated models. 
The model with the issue from maker lab is running for 15 minutes now and is still stuck at 15%. Suddenly it went up after 20 minutes until it completed the slicing process. I am happy that I can finally slice again with a newer version but to be honest with older versions I can slice this model in less than a minute.

AdamheesePRINTS, 2024-10-21:
Sorry I have to add another issue that I just faced after slicing a small very simple model. The size of the model is 70x50x57mm and when I hit slice an error appears that says: A g-code path exceeds the maximum size of printer height. When I scale it down by 98% it slices ok. When I scale it 101% it slices ok. It doesn't let me slice 99 and 100%.
[Spitzaufnahme PLA.zip](https://github.com/user-attachments/files/17464146/Spitzaufnahme.PLA.zip)