bambulab/BambuStudio #4956: undefined symbol: g_once_init_leave_pointer

tylerburleigh, 2024-10-07:
### Bambu Studio Version

1.09.07.52-24.04

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 23.10

### Printer

P1S

### How to reproduce

Launch AppImage from terminal: `./Bambu_Studio_ubuntu-v01.09.07.52-24.04.AppImage `

### Actual results

I get an error and it doesn't launch
`$ ./Bambu_Studio_ubuntu-v01.09.07.52-24.04.AppImage 
/tmp/.mount_Bambu_NPkWYA/bin/bambu-studio: symbol lookup error: /tmp/.mount_Bambu_NPkWYA/bin/bambu-studio: undefined symbol: g_once_init_leave_pointer`

### Expected results

I expect that it launches.

### Project file & Debug log uploads

No log file is produced (`$HOME/.config/BambuStudio/log` is empty after launching). 

No project file is open when the problem arises.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tylerburleigh, 2024-10-07:
The issue above occurs with `Bambu_Studio_ubuntu-v01.09.07.52-24.04.AppImage` but not `Bambu_Studio_linux_ubuntu-v01.09.01.66.AppImage`.

mprowe, 2024-10-08:
Same here!!
Not been able to get any version after 1.9.3 to work.
C'mon Bambu Labs. Please try harder.

MackBambu, 2024-10-09:
```
dpkg -l | grep libglib2.0*
```
Check the glib version. The 24.04 appimage is built on Ubuntu 24.04, so it may depend on a higher glib version.

mprowe, 2024-10-13:
Whooaaaa.....
Isn't the point of AppImages, that they should include ALL the REQUIRED resources so that it is OS agnostic!
Why are you skimping just to save a few kb of memory. Memory is cheap these days.

tylerburleigh, 2024-10-25:
@MackBambu 

```
dpkg -l | grep libglib2.0*
ii  libglib2.0-0:amd64                               2.78.0-2ubuntu0.1                         amd64        GLib library of C routines
ii  libglib2.0-0:i386                                2.78.0-2ubuntu0.1                         i386         GLib library of C routines
ii  libglib2.0-bin                                   2.78.0-2ubuntu0.1                         amd64        Programs for the GLib library
ii  libglib2.0-data                                  2.78.0-2ubuntu0.1                         all          Common files for GLib library
ii  libglib2.0-dev:amd64                             2.78.0-2ubuntu0.1                         amd64        Development files for the GLib library
ii  libglib2.0-dev-bin                               2.78.0-2ubuntu0.1                         amd64        Development utilities for the GLib library
```

What am I looking for?

crashkopf, 2024-11-11:
I'm having the same problem on Mint.  My glibc is v2.72.

The docs for the missing function say it was introduced in v2.80:
https://docs.gtk.org/glib/type_func.Once.init_leave_pointer.html

mprowe, 2024-11-11:
I don't know how or why, but [Bambu_Studio_linux_fedora-v01.10.00.81.AppImage](https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.81/Bambu_Studio_linux_fedora-v01.10.00.81.AppImage) is working for me!
I'm on openSUSE 15.6