bambulab/BambuStudio #4959: HiDPI Font scaling issues on Linux make the GUI unusable

phrxmd, 2024-10-08:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab github releases

### OS version

Linux (openSUSE Tumbleweed, KDE Plasma, Wayland, 200% display scaling)

### Additional system information

Operating System: openSUSE Tumbleweed 20240927
KDE Plasma Version: 6.1.5
KDE Frameworks Version: 6.6.0
Qt Version: 6.7.2
Kernel Version: 6.11.0-1-default (64-bit)
Graphics Platform: Wayland
Graphics Processor: Mesa Intel® UHD Graphics 620

This is on a notebook with a HiDPI display set to 200% display scaling in KDE.

### Printer

BambuLab X1 series

### How to reproduce

1. Start the application. That's it  

### Actual results

The GUI immediately comes up broken.

[image]

Some text-based UI elements are scaled outside their bounding boxes. Clicks are registered in the wrong location. The whole UI is hard to interact with, degrades and is in parts unusable.

[image]

The behaviour is the same with the AppImage from Github and with the Flatpak.

I have experimented with different combinations of `GDK_SCALE` and `GDK_DPI_SCALE`. With `GDK_SCALE` I can make the GUI larger and smaller overall. But the relative scaling of the text elements that flow outside their bounding boxes remains the same and is also unaffected by `GDK_DPI_SCALE`.

Starting with `GDK_BACKEND=X11` does not fix the issue.

The console is full of GTK errors:
```
> ./Bambu_Studio_linux_fedora-v01.09.07.52.AppImage
[2024-10-08 15:43:22.253718] [0x00007fcf94992a80] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
loading network module at /home/username/.config/BambuStudio/plugins/libbambu_networking.so
error, dlerror is /home/username/.config/BambuStudio/plugins/libbambu_networking.so: Kann die Shared-Object-Datei nicht öffnen: Datei oder Verzeichnis nicht gefunden
after dlopen, network_module is (nil)

(bambu-studio:555640): Gtk-CRITICAL **: 15:44:40.291: gtk_window_resize: assertion 'width > 0' failed

(bambu-studio:555640): GLib-GObject-CRITICAL **: 15:44:40.355: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:555640): Gtk-CRITICAL **: 15:44:40.355: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:555640): GLib-GObject-CRITICAL **: 15:44:40.356: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:555640): Gtk-CRITICAL **: 15:44:40.356: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:555640): GLib-GObject-CRITICAL **: 15:44:40.358: invalid cast from 'wxPizza' to 'GtkCellLayout'

[... 380 lines cut ...]
(bambu-studio:555640): Gtk-CRITICAL **: 15:46:23.732: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:555640): GLib-GObject-CRITICAL **: 15:46:23.735: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:555640): Gtk-CRITICAL **: 15:46:23.735: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:555640): GLib-GObject-CRITICAL **: 15:46:23.736: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:555640): Gtk-CRITICAL **: 15:46:23.736: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
```

### Expected results

The app should be usable. The GUI should look fine. There should be no scaling issues. 

### Project file & Debug log uploads

There is no project file because the behaviour is observable immediately, without loading a project.

Here are the logs: [logs.zip](https://github.com/user-attachments/files/17294300/logs.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

phrxmd, 2024-10-08:
Compare #1925, #2626. #3271, #3416.
Compare SoftFever/OrcaSlicer#886, SoftFever/OrcaSlicer#5171, SoftFever/OrcaSlicer#6513, SoftFever/OrcaSlicer#7050.

I think this is not an easy fix. But it would be nice to have at least a workaround, so that the app is usable.

MackBambu, 2024-10-09:
Does the new version of wxWidget support hidpi displays better? This seems to be a troublesome problem.

phrxmd, 2024-10-09:
For comparison: e.g. Audacity 3.6.4 uses wxWidgets 3.2.5 and its HiDPI behaviour is basically flawless. 

So it IS possible to build HiDPI-aware applications with wxWidgets.

ddmgh, 2024-12-07:
Any improved workaround for this? Have the same issue with Flatpak version 1.10.1.

mxtm, 2025-04-01:
Same issue here on Arch Linux running on a ThinkPad X1 Extreme Gen 3.

Using the workaround `GDK_SCALE=2 GDK_DPI_SCALE=0.5` suggested by @phrxmd helps, but it's still quite unusable.

[image]

Version `02.00.01.50-1` installed from the AUR package `bambustudio-bin`.

altendky, 2025-04-15:
[image]

I see issues here and there, but mostly the gui is usable.  The flushing volume shown here though is a bit tough.  This is with 2.0.1.50.

altendky, 2025-04-15:
[image]

FWIW, `GDK_DPI_SCALE=0.5 ./Bambu_Studio_ubuntu-20.04_v02.00.01.50.AppImage` got me to this teeny but otherwise closer to workable view.

lxe, 2025-08-06:
Yeah there's still a lot of stuff cut off and just missing from the UI