bambulab/BambuStudio #4962: Slicer generates incorrect G-code for a model.

Assimilator, 2024-10-09:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu P1S, Bambu X1C

### How to reproduce

Open project
Click "Slice plate"

### Actual results

Slicer generates incorrect G-code for layer 6, Steps 58769-58868.

### Expected results

correct G-code to print the model

### Project file & Debug log uploads

[6x6 Bracket - Bolt-Locked Inserts.zip](https://github.com/user-attachments/files/17300889/6x6.Bracket.-.Bolt-Locked.Inserts.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

ttt888zzz, 2024-10-11:
HI [Assimilator](https://github.com/Assimilator)：This issue has been fixed internally, please wait for  the update in the near future

ttt888zzz, 2024-10-16:
The latest version has been fixed.
 Please visit the website  at https://github.com/bambulab/BambuStudio/releases and try again.

closed 2024-10-16 by ttt888zzz