bambulab/BambuStudio #4966: BambuStudio unable to download network plugin

thrasher, 2024-10-09:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.0 (24A335)

### Additional system information

2024 Apple MacBook Air
Chip: Apple M3
Memory 16 GB

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1. Open BambuStudio app
2. Observe: "Downloading Bambu Network Plugin" dialog automatically opens.
3. Observe: network plugin dialog remains at 0% progress
4. Observe: dialog message: "Failed to download the plug-in."
5. Unable to use networking in BambuLabs Studio (can't login, see printers, or anything).
6. Help / Open Network Test / Start Test Single-Thread fails (see screenshot)

### Actual results

Network test fails (the network plugin can't be downloaded).

[image]

### Expected results

The software shouldn't need to install a plugin to access the network.

### Project file & Debug log uploads

N/A: This failure is not for a 3mf file.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

thrasher, 2024-10-09:
Also note:
iCloud Private Relay is turned "off"
Cisco AnyConnect is not installed
AVG Security is not installed

These were referenced in the wiki here: https://wiki.bambulab.com/en/software/bambu-studio/failed-to-get-network-plugin

lanewei120, 2024-10-14:
@thrasher 
could you show a full picture of the test results, the picture you attached previously just shows the testing begins
we can not see the testing results

and also, please help to upload the logs to us for help debugging

from our experience, it seems bambu studio is blocked to access network
could you also check from this from your system

gerases, 2025-07-19:
Does anyone know what the url to the plugin is on a mac? the path returned when using curl and the url mentioned in the docs returns a reference to a file for windows.

gerases, 2025-07-19:
This worked: curl -O "https://public-cdn.bblmw.com/upgrade/studio/plugins/02.01.01.52/30f50b382e/mac_02.01.01.52.zip"

You then need to put the contents of that archive into `~/Library/Application Support/BambuStudio/plugins`