bambulab/BambuStudio #4968: bambu studio missing 3d view

hb-hiber, 2024-10-09:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

window 10

### Additional system information

CPU:Intel(R) Core(TM) i7-7700K CPU @ 4.20GHz   4.20 GHz
GPU:NVIDIA GeForce GTX 1060 6GB

### Printer

Bambu Lab A1

### How to reproduce

1.install bambu studio
2.start bambu studio

### Actual results

missing 3D view ：
[image]
[image]

### Expected results

other machine exists 3d view ：
[image]

### Project file & Debug log uploads

[BambuStudio.zip](https://github.com/user-attachments/files/17317466/BambuStudio.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Haidiye00, 2024-10-10:
i suggest you try upgrading the Nvidia driver

hb-hiber, 2024-10-12:
> i suggest you try upgrading the Nvidia driver

Thank you for your suggestion.
My driver is already the latest version, but the 3D view is still missing

conq, 2024-10-25:
I have the exact same issue. I have an Nvidia RTX 2070 running driver version 560.35.03 on Ubuntu 24.10.

conq, 2024-10-25:
I do get a lot of these error messages in the log:
(bambu-studio:147077): Gtk-CRITICAL **: 18:32:49.718: gtk_widget_set_size_request: assertion 'width >= -1' failed

and 

(bambu-studio:147077): Gtk-CRITICAL **: 18:32:42.805: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

Haidiye00, 2025-03-26:
Hello, we have released software V2.0. Please download and use it
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.00.95

closed 2025-03-26 by Haidiye00