bambulab/BambuStudio #4969: Reload STL resets location and orientation

replaysMike, 2024-10-09:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab X1 Carbon

### How to reproduce

This was previously reported as fixed/cannot reproduce in issue #910 but it does indeed still occur, as of version 1.9.7.52 - requesting this gets reopened. The reason its an issue is it doesn't reorient all the modifiers and support enforcers so now they are all in the wrong location/orientation.

I should note this doesn't always happen, I'm not sure what causes it to occur. STL's are edited/generated from Blender, with no change in units, orientation or scale.

Before reload:
[image]

After reload:
[image]

### Actual results

Orientation is flipped on reload

### Expected results

No change in orientation or location

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/17319904/log.zip)
[TopShelf-Right.zip](https://github.com/user-attachments/files/17319905/TopShelf-Right.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

replaysMike, 2024-10-09:
Worth noting, closing and re-opening Bambu Studio caused the issue to go away. I was able to reload the model and it kept its orientation this time. So definitely a bug here

Haidiye00, 2024-10-10:
Can you provide the original STL file and record the process of reproducing it? Thank you

replaysMike, 2024-10-17:
> Can you provide the original STL file and record a gif for the process of reproducing it? Thank you

I was able to reproduce it again, here's the gif & files.

[image]

[STLFiles.zip](https://github.com/user-attachments/files/17422911/STLFiles.zip)
* Zip contains the project file, previous model, and replacement model (though it's not the model, restarting Bambu Studio fixes the problem)

Haidiye00, 2024-10-18:
[image]
Thank you for your discovery. We have identified the issue. thank you

Haidiye00, 2024-10-29:
Hello, 1.10 beta2 has been released. Please download and use it. Thank you~

closed 2024-10-29 by Haidiye00