bambulab/BambuStudio #4974: Bambu Studio not rendering slicer window on Fedora

pypeaday, 2024-10-10:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab github releases

### OS version

Fedora 40

### Additional system information

GPU: Nvidia 3090

### Printer

A1

### How to reproduce

I imagine you install Fedora 40, install the flatpak or AppImage (I've tried both), open the application but then have the slicer window never render

### Actual results

When I open my app the sliver window never renders

### Expected results

I would expect to see my model so that I can slice and print it

### Project file & Debug log uploads

I hopened some logs and this looked like a relevant error

[debug_Thu_Oct_10_08_08_44_2.log.0.zip](https://github.com/user-attachments/files/17328658/debug_Thu_Oct_10_08_08_44_2.log.0.zip)

```
2024-10-10 08:02:53.498474[Thread 0x00007f3be354a3c0]:LoadProfileFamily: parse /app/share/BambuStudio/profiles/Qidi.json got a nlohmann::detail::parse_error, reason = [json.exception.parse_error.101] parse error at line 1, column 1: syntax error while parsing value - unexpected end of input; expected '[', '{', or a literal
```

### Checklist of files to include

- [X] Log file
- [ ] Project file

MackBambu, 2024-10-11:
https://github.com/bambulab/BambuStudio/issues/4669
Downgrading to nvidia 555

JaimeHW, 2024-10-13:
Downgrading the Nvidia driver did not work. This bug has made BambuStudio entirely unusable for us on linux.

ptillemans, 2024-10-23:
Same issue on Arch Linux. Starting the app image with 

```
#!/bin/bash

APPIMAGE=~/.local/bin/Bambu_Studio_ubuntu-v01.09.07.52-24.04.AppImage

__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json WEBKIT_DISABLE_DMABUF_RENDERER=1 WEBKIT_FORCE_COMPOSITING_MODE=1 WEBKIT_DISABLE_COMPOSITING_MODE=1 $APPIMAGE
```

works fine till this issue is fixed.

MackBambu, 2024-11-18:
merge issue: https://github.com/bambulab/BambuStudio/issues/4669

closed 2024-11-18 by MackBambu