bambulab/BambuStudio #4975: Flathub verification

secretmango, 2024-10-10:
[image]

Reading your README, I understood that you build the Flatpak, which is great!

Could you do a Flathub verification process, to remove that dangerous sign? There are also people using the `subset=verified` Flathub repo, which would only then get your app displayed.

The docs on how to do verification [are here](https://docs.flathub.org/docs/for-app-authors/verification#im-publishing-an-app-on-flathub-how-do-i-get-it-verified) and it looks like simply some app author needs contributor rights to the [Flathub Github repo](https://github.com/flathub/com.bambulab.BambuStudio) which is maintained by @hadess it seems, so this should work.

hadess, 2024-10-10:
> The docs on how to do verification [are here](https://docs.flathub.org/docs/for-app-authors/verification#im-publishing-an-app-on-flathub-how-do-i-get-it-verified) and it looks like simply some app [...]

Both @lanewei120 and @MackBambu already have access to the Flathub repo. I provided them with the information about verification back in January. Hopefully some work will be done on this soon, it's such a shame to miss out on verification when all the authorisations are already here.

MackBambu, 2024-10-11:
[image]
The verification has just been completed

closed 2024-10-11 by MackBambu