bambulab/BambuStudio #4976: Wrong Filament Usage

Parceval4482, 2024-10-10:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

P1P

### How to reproduce

I have several printing plates in the Bambustudio and in the preview tab on the All Plates Stats the total amount of filament used is not correct. 2 pieces are there. Filament 1 is support filament and filament 2 is for the print object. 

In the total amount required section, the same amount is shown for the supports as for the model.

In my case: 
Total 
Model: 367.99g and 121.42m 
Supports: 367.99g and 121.42m

Supports filament 1 + 2:
added together by me: 10.32g+36.3g = 46.62g <<< That should actually be in Total.

This also results in a total filament requirement of 845.58g and not 414.61g

Translated with DeepL.com (free version)

### Actual results

[image]
[image]

### Expected results

Correct Filament usage

### Project file & Debug log uploads

[Stargate first Parts Plate 5-8.zip](https://github.com/user-attachments/files/17331797/Stargate.first.Parts.Plate.5-8.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ttt888zzz, 2024-10-11:
I'm Sorry.The value of the support material is not displayed correctly, the internal version has been fixed. A new version will be released in the near future.

ttt888zzz, 2024-10-16:
The latest version has been fixed.
 Please visit the website  at https://github.com/bambulab/BambuStudio/releases and try again.

closed 2024-10-16 by ttt888zzz